jgarff/rpi_ws281x #350: ws2811_init fails with error code -3 on RPi 3B+ with ArchLinuxARM aarch64

nils-van-zuijlen, 2019-02-03:
I am using the python library at [rpi-ws281x/rpi-ws281x-python](https://github.com/rpi-ws281x/rpi-ws281x-python) and I recently switched from raspbian to ArchLinuxARM aarch64 on my RPi.

After the upgrade, when I tried to run my code, I got
```text
Traceback (most recent call last):
  File "LedPanel.py", line 129, in <module>
    panel = LEDPanel(0,1)
  File "LedPanel.py", line 22, in __init__
    self._strip.begin()
  File "/usr/lib/python3.7/site-packages/rpi_ws281x/rpi_ws281x.py", line 124, in begin
    raise RuntimeError('ws2811_init failed with code {0} ({1})'.format(resp, str_resp))
RuntimeError: ws2811_init failed with code -3 (Hardware revision is not supported)
```

As I wanted to know what happened, I traced the calls back to the C library in which `ws2811_init` was the function called and the error code was thrown [at line 886](https://github.com/jgarff/rpi_ws281x/blob/master/ws2811.c#L886).

Apparently, the return value of `rpi_hw_detect` was wrong. I later found that it was trying to read `/proc/device-tree/system/linux,revision` from my system ([code at line 343 of rpihw.c](https://github.com/jgarff/rpi_ws281x/blob/master/rpihw.c#L343)) and that that file does not exist on my Pi.

How may I get it to work ?

nils-van-zuijlen, 2019-02-05:
PS: I think #183 may fix the problem I have (and somebody else may have at some time)

jamesshawver1, 2020-02-21:
I think I found the issue with this.
Patch needed in rpihw.c - the detection for ARM64 isn't working due to the size of detection code. fread returns the number of elements, not the number of bytes read, when the guarding statement is comparing against the number of bytes read.  See - https://www.tutorialspoint.com/c_standard_library/c_function_fread.htm

see line number 419/420 -  

419     size_t read = fread(&rev, sizeof(uint32_t), 1, f);
420     if (read != sizeof(uint32_t))

should be

419     size_t read = fread(&rev, sizeof(uint32_t), 1, f);
420     if (read != 1)

fooker, 2021-06-16:
I have the same problem even with #395 applied (which does exactly what was mentioned in https://github.com/jgarff/rpi_ws281x/issues/350#issuecomment-589457029). The problem is still that the file `/proc/device-tree/system/linux,revision` is not there on my system. In fact, the whole `/proc/device-tree/system/` is missing.

I'm running on `linux-5.4.106` with `broadcom/bcm2837-rpi-3-b.dtb` loaded.

Some maybe helpful thing i found:
```
# uname -a
Linux photonic 5.4.106 #1-NixOS SMP Wed Mar 17 16:03:58 UTC 2021 aarch64 GNU/Linux
# cat /proc/device-tree/model 
Raspberry Pi 3 Model B
# cat /proc/device-tree/serial-number 
00000000b43e053c
# cat /proc/device-tree/compatible 
raspberrypi,3-model-bbrcm,bcm2837
```

closed 2022-01-25 by Gadgetoid