jgarff/rpi_ws281x #353: Compute Module 3 PWM Errors

pablogs9, 2019-03-08:
I'm addressing 3 WS2812B using a RPi CM3 on pin 14 (PWM0 expected).
When I capture the signal with the oscilloscope, it seems to be a clock burst of 2.14 ms aprox.
The 800 kHz frequency is OK, the high and low time for 0 and 1 symbols are correct (same as datasheet). I expected 1.25 us * 8 bits * 3 colors * 3 LEDS = 90 us burst, but it seems to send more data.

I'm running the ./test compilation of this project and I have taken into account that it refresh every 1/15 s (the bursts are separeted enought).

Thanks in advice.

pablogs9, 2019-03-08:
PD: supply is 3V3 and interfacing signal DIN is also 3V3. This is a voltage configuration that is working for me on other hardware.

jgarff, 2019-03-29:
I'm sorry, but you really need to convert the signal to 5V for things to work properly and reliably.  People get lucky when running at 3v3, but there's no guarantee.  You're right on the edge of a 0-1 transition on the 5V side with a 3v3 driver.  Just because it works on another device is not enough.  Every electronic component is different and has known variation, which is why you need a level translator.

closed 2019-03-29 by jgarff

pablogs9, 2019-03-29:
I think that electrical operation of our system if fully guaranteed, just asking about a possible timing/protocol bug on the PWM generation of burst because of it unsual length... 

Thanks anyway