jgarff/rpi_ws281x #354: Synchronization mechanism?

davthomaspilot, 2019-03-24:
It seems the hardware_render call is asynchronous.  That is, the call returns before data transfer to the WS281X strips has completed.

The hardware I'm using gates the the data sent to the strips.  So, different data can be sent to strips not directly connected.

So, it's necessary for my code to wait until transfer to one strip has completed before gating off the data to that strip and starting transfer to another.

I can add delays to accomplish what I need, but the delay needed will be a function of the number of LEDs in the strip being driven.  It would be better to have a way to block execution until the DMA (or other) transfer has completed.

Has anyone thought about how this might be done?  Thought I'd asked before digging in and subsequently being told "been there, done that".

Any suggestions for how to do this, eg. add a callback that gets executed when transfer has completed, or use of mutex that the caller can optionally wait for, etc.

davthomaspilot, 2019-03-24:
BTW, I'm using PWM on an rpi3B+.

I conclude that hardware_render call returns before the dma transfer has completed be timing how long it takes to execute the hardware_render call.

I'm seeing times as low as 110 usec.  That's driving a 10 LED strip at 800 khz.  That should take 400 usec:

> Frequency                              800 khz
> Time per bit (usec)                1.25
> Time per LED  (32 bits)         40 usec
> LEDS per strip                      10
> Delay one   side (msec) |     .4

Maybe the code is SUPPOSED to block until the DMA transfer is complete?   I see a function ws2811_wait that loops waiting for an active flag to clear (or and error to occur):

    while ((dma->cs & RPI_DMA_CS_ACTIVE) &&
           !(dma->cs & RPI_DMA_CS_ERROR))
    {
        usleep(10);
    }

So,  will RPI_DMA_CS_ACTIVE stay true until the dma transfer (to the LED strip) has completed?

That flag is part of dma_t structure.  Any pointers about how to associate that with a hardware spec would also be appreciated.

jgarff, 2019-03-24:
Correct, the RPI_DMA_CS_ACTIVE flag remains high until the hardware DMA
operation is complete.  If you want to block your code until the DMA
transfer is done, use the ws2811_wait call.  That's exactly why it's
there.  :)

davthomaspilot, 2019-03-24:
When I said "hardware_render", I meant ws2811_render (hardware_render is just a wrapper for that function in my code).

But ws2811_render already calls ws2811_wait.  But clearly, it's not waiting long enough, based on timing of the call and other issues I see when I change a steering device immediately after ws_2811_call returns.

So, it sounds like the intended behavior is to block until the dma completes?  That would work, but it would be better to have it return immediately.  That way, code could execute in parallel until it must block to avoid an issue.  In my case, I'd block waiting for completion of the DMA at the point I'd change the "data steering" to a different LED strip.

I haven't updated my ws2811x code to the latest since last fall, so first thing I'll do is get the latest.  Then, I'll add some print statements in teh ws2811x code so others can reproduce the issue (if it isn't eliminated with newer code).

After that, I may need help understanding more about the signal that's used to indicate completion of data transfer.

davthomaspilot, 2019-03-24:
Updated to the latest ws2811x code and still see the same issue.  I'll add some prints of the timestamps to show ws2811_render completes sooner that what it would take to transfer the data to the LED strip(s).

davthomaspilot, 2019-03-24:
Looking at the code again, I see comments in ws2811_render code immediately before the wait:

// Wait for any previous DMA operation to complete.

So, I now realize that wait is before the dma operation is inititated.  So, it takes another call to the wait function to block until the operation completes.

Perfect!  (If I go this right).

Thanks for the reply (and I'd appreciate confirmation I got it right).

davthomaspilot, 2019-03-24:
I added a wait call, and now the execution time is what I'd expect.

This seems like an important interface to know about.  You might want to mention it in the readme documentation.

I'd do it, if that's what's expected here.  I'm not sure what it takes to get authorized to make changes.

closed 2019-03-24 by davthomaspilot

jgarff, 2019-03-24:
Yep, exactly.  I didn't want the user to pay the CPU cost when they could
be doing something in parallel.  As you noticed, the stall is just there
before to wait until the DMA is ready to work again.