jgarff/rpi_ws281x #358: No module named 'NEOPIXEL'

kinjalsoft, 2019-04-25:
Hi..
So basically I recently bought a Raspberry pi with AlphaBot2 pi 
The issue is that when I try to run the Line Follower program from the zip provided in the manual there is an error which comes and that is : 
Import error *
No module named "neopixel"
I have also attached one script with it 
[Line Follow.txt](https://github.com/jgarff/rpi_ws281x/files/3117962/Line.Follow.txt)

Please answer this issue

ChanJeff123, 2019-05-20:
copy example code to rpi_ws281x-master/python file

Gadgetoid, 2019-06-28:
Change `from neopixel import *` to `from rpiws281 import *`. The "neopixel" name was conflicting with an Adafruit package, and since it's their trademark it was dropped entirely from the shipped Python library.

Original discussion: https://github.com/rpi-ws281x/rpi-ws281x-python/issues/8
And notice that it was dropped: https://github.com/rpi-ws281x/rpi-ws281x-python/issues/10

combs, 2020-05-12:
I have a PR in for this, #399