jgarff/rpi_ws281x #364: Hardware revision is not supported

karcio, 2019-07-02:
HI, I have the same problem like issue #342 

Trying to run RotorHazard race tracker on raspberry pi 3
pi@delta5:~/RotorHazard/src/server $ python server.py
Configuration file imported
Node 1 found at address 8
Node 2 found at address 10
Node 3 found at address 12
Node 4 found at address 14
No node at address 16
No node at address 18
No node at address 20
No node at address 22
Node 1: API_level=14, Freq=5760, EnterAt=250, ExitAt=240
Node 2: API_level=14, Freq=5695, EnterAt=192, ExitAt=160
Node 3: API_level=14, Freq=5732, EnterAt=192, ExitAt=160
Node 4: API_level=14, Freq=5917, EnterAt=192, ExitAt=160
Language file imported
Traceback (most recent call last):
File "server.py", line 279, in <module>
strip.begin()
File "build/bdist.linux-armv7l/egg/neopixel.py", line 106, in begin
RuntimeError: ws2811_init failed with code -3 (Hardware revision is not supported)

I found that solution can be resolved as closed issue #342 but I have latest code and still this is happen.

Gadgetoid, 2019-07-02:
You need to specify your hardware revision- a specific model of Pi (ie: 3B+) could have several revisions for reasons as simple as being manufacturered in a different location.

Run `cat /proc/cpuinfo | grep Revision` and paste the results here.

Also, if you're using `neopixel.py` (the Python examples in this repository) you're running deprecated, largely unsupported code. You should migrate to the Python bindings - https://github.com/rpi-ws281x/rpi-ws281x-python

karcio, 2019-07-02:
Hi @Gadgetoid 
Sure, sorry about that
pi@delta5:~ $ cat /proc/cpuinfo | grep Revision
Revision	: a02083

So I cloned repo you point me then 
pi@delta5:~ $ sudo pip uninstall rpi_ws281x
Uninstalling rpi-ws281x-1.0.0:
  Would remove:
    /usr/local/lib/python2.7/dist-packages/rpi_ws281x-1.0.0-py2.7-linux-armv7l.egg
Proceed (y/n)? y
  Successfully uninstalled rpi-ws281x-1.0.0
pi@delta5:~ $ sudo pip install rpi_ws281x
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting rpi_ws281x
  Downloading https://files.pythonhosted.org/packages/e6/2e/ec156bf6e0f2288a18fc148cc5b985f6eb2e2bb202859046e633e841c636/rpi_ws281x-4.2.1-cp27-cp27mu-linux_armv7l.whl (102kB)
    100% |████████████████████████████████| 112kB 818kB/s 
Installing collected packages: rpi-ws281x
Successfully installed rpi-ws281x-4.2.1

but now I can see other issue 
pi@delta5:~/RotorHazard/src/server $ python server.py 
Traceback (most recent call last):
  File "server.py", line 28, in <module>
    from neopixel import *
  File "/usr/local/lib/python2.7/dist-packages/gevent/builtins.py", line 96, in __import__
    result = _import(*args, **kwargs)
ImportError: No module named neopixel

Gadgetoid, 2019-07-03:
The `neopixel` module has been dropped. Use `from rpi_ws281x import *` instead.

Your Revision ID is awful close to the one merged here -https://github.com/jgarff/rpi_ws281x/pull/362 - and is, in fact, not currently supported by the codebase so rpi-ws281x will not work for you at all. Yet.

I'll get this Revision ID merged and another point release out, since there are a couple of changes to go with it.

Gadgetoid, 2019-07-03:
Support should be available in versions >= 4.2.2

```
sudo pip install --upgrade rpi_ws281x
```

karcio, 2019-07-03:
Hi @Gadgetoid 
Thank You very much for this quick commit. Working fine now. Cheers

closed 2019-07-03 by karcio