jgarff/rpi_ws281x #369: Code not running on 3A+

danaswift, 2019-07-24:
I'm having a problem similar to #364. 

I'm trying to run this code:
```
import board
import neopixel
import time

pixels = neopixel.NeoPixel(board.D10, 16)
pixels[1] = (255,0,0)

```
For a strip of neopixels but keep getting this error

> Traceback (most recent call last):
>   File "./neopixeltestcode.py", line 8, in <module>
>     pixels[1] = (255,0,0)
>   File "/usr/local/lib/python3.5/dist-packages/neopixel.py", line 175, in __setitem__
>     self.show()
>   File "/usr/local/lib/python3.5/dist-packages/neopixel.py", line 230, in show
>     neopixel_write(self.pin, self.buf)
>   File "/usr/local/lib/python3.5/dist-packages/neopixel_write.py", line 24, in neopixel_write
>     return _neopixel.neopixel_write(gpio, buf)
>   File "/usr/local/lib/python3.5/dist-packages/adafruit_blinka/microcontroller/bcm283x/neopixel.py", line 65, in neopixel_write
>     raise RuntimeError('ws2811_init failed with code {0} ({1})'.format(resp, message))
> RuntimeError: ws2811_init failed with code -3 (Hardware revision is not supported)
> swig/python detected a memory leak of type 'ws2811_t *', no destructor found.

It works with no errors on a RPI 3B but I can't get it to work with the 3A+. It's hardware revision 9020e0. Any pointers in the right direction would be appreciated!

pjenns, 2019-09-29:
Same issue here with a Pi 3 A+. Finally got three builds done for a small contracted project and was rebuilding the SD cards from scratch and found that the Neopixels no longer work.

pjenns, 2019-10-01:
Fixed by installing a previous release of ws281x.
`sudo pip3 install rpi_ws281x==4.1.0`

Gadgetoid, 2019-11-06:
This seems to be a duplicate of #342 which was fixed by me in https://github.com/jgarff/rpi_ws281x/commit/0d99943da134b9db368994c4ccc7a2e1cb109d4d and is even still present in `rpihw.c` here: https://github.com/jgarff/rpi_ws281x/blob/6b01e53c3f8bbcfd2c49c447d4c6a6e2b0031370/rpihw.c#L358-L364 and a quick check of the source for `rpi_ws281x==4.2.2` also shows it.

Since there hasn't been a regression here, I'm wondering if this is somehow related to this Revision ID being the third to last in the list of supported devices in `rpihw.c`.

Gadgetoid, 2019-11-06:
Oh- my bad! It looks like you were accidentally running Adafruit's code instead of `rpi_ws281x`. The `rpi_ws281x` library does not provide `neopixel` as of version 4.0.0 (not sure why 4.1.0 works for you!). See: https://github.com/rpi-ws281x/rpi-ws281x-python/releases/tag/v4.0.0

You should be using `import rpi_ws281x` instead of `import neopixel`

closed 2019-11-06 by Gadgetoid