jgarff/rpi_ws281x #372: Raspberry pi 4 support

b4nter, 2019-10-15:
Got this model https://www.waveshare.com/wiki/RGB_LED_HAT

This RGB_LED_HAT does not respond with Raspberry Pi 4 Model B Rev 1.1  after running test program. 

I can confirm LED HAT itself is fully working as I tested it on my colleagues Raspberry Pi 2 with the same software and it was working all fine.

Have tried to run on different GPIO's no luck with that.

Can't tell for sure, but it might be missing some of hardware detection in file file rpihw.c
```
--
48 | #define VIDEOCORE_BASE_RPI                       0x40000000
49 | #define VIDEOCORE_BASE_RPI2                      0xc0000000
50 |  
51 | #define RPI_MANUFACTURER_MASK                    (0xf << 16)
…  |  
62 | .videocore_base = VIDEOCORE_BASE_RPI2,
63 | .desc = "Pi 4 Model B - 1GB"
64 | },
65 | {
66 | .hwver = 0xB03111,
```
SDRAM for this model is 0x1800000 got it by running bcm_host_get_sdram_address()
reference: https://www.raspberrypi.org/documentation/hardware/raspberrypi/peripheral_addresses.md

Tried to change some variables around, but still can't get it working on my RPI4. Was wondering if anyone else tried to get this RGB_LED_HAT working on RPI4. Or someone more experienced can have a look at this issue.

Gadgetoid, 2019-11-06:
I'm unfamiliar with RGB LED HAT, but I confirmed our - very similar - Unicorn HAT product as working on the Pi 4 and haven't seen any reports otherwise. Have you managed to get it up and running since?

b4nter, 2019-11-07:
Unfortunately no, tried with the most recent update too.

2E0PGS, 2020-01-26:
## A bit more information.

I know @b4nter IRL.

We both have been looking at this issue.

I have had @b4nter waveshare neopixel (WS2812B) hat working on my PI 2B with the C test program and Python examples.

We couldn't get it to work on his `Raspberry Pi 4 Model B Rev 1.1`. Even after updates to the code for hardware detection of that model.

Oddly enough I have quite a lot of SMART strip WS2182B LEDs and I tried them with the C test program and they work on both PI's fine. However I only use a test strip of 4 pixels. Perhaps there is a timing issue.

We tested different forced timings in the code. We also tested only lighting one pixel. None of this worked for the PI4 and that hat.

I will try measure the PWM rates of each board with the test program and a scope. However I need to aquire or borrow a PI4 for that.

PI4 is using official PSU.

## Pictures of the test strip that did work on PI4

[image]

[image]

jgarff, 2020-01-26:
Are you using proper voltage level translators?  Running 3v3 signals are
not reliable and will cause intermittent problems and/or hardware and
temperature variations.  Working fine on some hardware and not on others is
exactly what I would expect if you're not translating the voltage
properly.  Please see the wiki pages for more information.

2E0PGS, 2020-01-26:
The board is designed for PI use. It even has examples using this library.

I am not sure if it has logic level converters on board. It looks like does not on my physical inspection.

I just did a bit of googling and found its schematic: https://www.waveshare.com/w/upload/b/bf/RGB_LED_HAT_Schematic_.pdf

No transister logic level converters. I can check the logic levels on a scope later. Perhaps the PI2 is a slightly higher 3.3V due to less power consumption of onboard components.

The LED tapes I am using are not using logic level converters however they work fine on both devices so ignore those. We know they work.

jgarff, 2020-01-26:
Yeah, that board clearly needs a level shifter and was designed without
looking at the specs for either the ws281x or Pi.  The highest output of a
Pi I/O is ~3.6, and nominal is 3.3, but can be lower.  The ws2812 is
Vdd(5v) - 0.7, or 4.3V for a logic level high.  So this board is clearly
out of spec.  Most people are lucky in that it works, but chip to chip
variation, temperature, etc., is a problem.

See the wiki page on the project for a example Pi hat that I designed which
includes a level shifter.

2E0PGS, 2020-01-26:
Yes originally the issue was missing support for his specific PI4 hardware revision detection. So there was a software issue there. That has since been merged.

But now we got this far it does indeed seem it's of poor design now I found the shematic which I didn't find before.

I can build a logic level converter and/or still check we have no timing issues since that was a problem at one point in time. Then we can put this to bed.

I think @b4nter said the board is no longer listed for sale? Is that correct @b4nter ?

2E0PGS, 2020-01-26:
Anyway,

Thank you for the reply and comment's.

Logic level is something I checked loosely with a meter before, however I haven't had time to hook up a scope to it. PWM frequency would effect my testing previously given the hardware detection I mentioned before.

Logic level probably is the cause now we gotten past previous issues. The issue is trying to make it work. Either he buys a PI2 or we try a different PSU and hope the tolerances fit.

Or he buys a new board as you suggested.

I'll get it working one way or another and report back just for closure.

Cheers,

⁣-- 
Peter Stevenson (2E0PGS)
http://www.m3pgs.co.uk​

jgarff, 2020-01-26:
Sounds good.  I wouldn't rely on getting a Pi2 to solve the
problem though.  It may work, but may not.  Chances are probably it will
just based on the number of people who find success.  Just remember, it's
not the Pi4 at fault.  A Pi2 could also not product 3v6 (or 3v3) on the I/O
and still be within spec of the Pi.  Any use of any Pi with any WS281X
without a level shifter is out of spec.  Getting another Pi to solve the
problem is just rolling the dice.  Also keep in mind that even if a Pi
works today, it might not tomorrow.  Remember temp/voltage/power
supply/etc., also are a factor.

2E0PGS, 2020-01-26:
Yeah I didn't mention that as it's a given.

Well aware.

⁣-- 
Peter Stevenson (2E0PGS)
http://www.m3pgs.co.uk​

jgarff, 2020-01-26:
Sounds good.  Good luck.

penfold42, 2020-01-26:
> The  ws2812 is Vdd(5v) - 0.7, or 4.3V for a logic level high. 

I thought it was 0.7 *  Vdd = 3.5v

2E0PGS, 2020-01-26:
I personally assume they're expecting a decent signal given its 5v. So I would consider it lucky to have it working on a PI at all given its logic level 3.3v nominal.

However if you happen to know the specific bit bang high threshold is 0.7 then it calcs to

```

jgarff, 2020-01-26:
Oh good catch penfold, thanks.  I believe you're right.  I think the
problem is still that we're out of spec, but the margins are much less than
I had originally described, which probably describes why it works most of
the time.

Thanks,

Jeremy

2E0PGS, 2020-01-26:
I personally have ran I2C between PI and Ardunio without logic level converters with sucess. 

Given I only did it one direction. Bi-directional would destroy the PI GPIO. 

Anyway that was with a PI2 B and a PI Zero W. 

So I suspect with the added onboard power consumption of a more powerful CPU of the PI4 and different power circuitry (USB C) it probably has a different GPIO logic level in real world measurements.

2E0PGS, 2020-01-26:
So please let me scope it out to see what the real numbers are and differences to prove why it works on a PI 2B and not a PI 4. Regardless of it's debatable design.

jgarff, 2020-01-26:
I2c is a different animal entirely.  Pull-ups are present on the bus to
drive the signal high.  Each entity then pulls it low.  Additionally, and
arduino (typically AVR or such) has protection diodes, which drops the io
level to vcc + ~0.7.  I'd just be careful here making this comparison.
It's not apples to apples.  FWIW, I still believe you're on the margins of
functionality and this is still very likely a voltage level issue.

2E0PGS, 2020-01-27:
I wouldn't rely on a built-in input protection for that. Opto-isolated or transistor based logic level converter, external protection diodes or external resistors is really the only way to not blow it up. However I was referring to PI as the master and Arduino as a slave device so it's safe. Otherway around PI needs protection.

However let's not get off topic.

2E0PGS, 2020-01-31:
## My PI 2B with Waveshare hat connected via breadboard 5v, gnd, p18

[image]

## @b4nter PI 4 with Waveshare hat connected via breadboard 5v, gnd, p18

[image]

## Time/DIV

[image]

## Volts/Div

[image]

## Summary

Both worked when hooked up via breadboard to those specific pins. So I suspect the hat uses a different pinout (uses a different ground) which is faulty pin on @b4nter 's PI 4. They also both worked on 3.3V supply too.

closed 2020-01-31 by b4nter

2E0PGS, 2021-02-06:
Hi all, I finally bought a Pi 4 B 4 GB just to confirm there was something wrong with the pins on @b4nter Pi. The above scope tests already proved the logic levels and frequency are the same between my working Pi 2 B and his Pi 4 B 4 GB so here I am just further confirming it with a new board. I am using the original SD card from many months ago to rule out software configuration or version differences.

```
$ cat /proc/cpuinfo
...
Hardware	: BCM2835
Revision	: c03112
...
Model		: Raspberry Pi 4 Model B Rev 1.2
```

Some `.bash_history` for others sake

```
sudo apt install git
git clone https://github.com/jgarff/rpi_ws281x.git
cd rpi_ws281x/
sudo apt install scons
```

```
~/rpi_ws281x $ git log
commit 6851d9fb090f8a4703d2ceac97da2de617b09e8d (HEAD -> master, origin/master, origin/HEAD)
Author: Stephen Warren <swarren@wwwdotorg.org>
Date:   Tue Aug 27 09:24:18 2019 -0600
```

```
~/rpi_ws281x $ git status
On branch master
Your branch is up to date with 'origin/master'.

Changes not staged for commit:
  (use "git add <file>..." to update what will be committed)
  (use "git checkout -- <file>..." to discard changes in working directory)

	modified:   main.c
	modified:   rpihw.c

no changes added to commit (use "git add" and/or "git commit -a")
```

```
~/rpi_ws281x $ git diff
diff --git a/main.c b/main.c
index f88414a..e0d0894 100644
--- a/main.c
+++ b/main.c
@@ -59,8 +59,8 @@ static char VERSION[] = "XX.YY.ZZ";
 #define TARGET_FREQ             WS2811_TARGET_FREQ
 #define GPIO_PIN                18
 #define DMA                     10
-//#define STRIP_TYPE            WS2811_STRIP_RGB               // WS2812/SK6812RGB integrated chip+leds
-#define STRIP_TYPE              WS2811_STRIP_GBR               // WS2812/SK6812RGB integrated chip+leds
+#define STRIP_TYPE            WS2811_STRIP_RGB         // WS2812/SK6812RGB integrated chip+leds
+//#define STRIP_TYPE              WS2811_STRIP_GBR             // WS2812/SK6812RGB integrated chip+leds
 //#define STRIP_TYPE            SK6812_STRIP_RGBW              // SK6812RGBW (NOT SK6812RGB)
 
 #define WIDTH                   8
diff --git a/rpihw.c b/rpihw.c
index 941437c..fd77818 100644
--- a/rpihw.c
+++ b/rpihw.c
@@ -70,7 +70,7 @@ static const rpi_hw_t rpi_hw_info[] = {
         .desc = "Pi 4 Model B - 2GB"
     },
     {
-        .hwver = 0xC03111,
+        .hwver = 0xC03112,
         .type = RPI_HWVER_TYPE_PI4,
         .periph_base = PERIPH_BASE_RPI4,
         .videocore_base = VIDEOCORE_BASE_RPI2,
 ```

```
~/rpi_ws281x $ scons
~/rpi_ws281x $ sudo ./test
```

[image]

[image]

All working.