jgarff/rpi_ws281x #373: Any chance of adding support for LPD8806-Strips?

Snille, 2019-10-17:
Hi! I'm sitting on a bunch of LPD8806-Strips... It's also a SPI-based controlled strip. Is there any chance that you may add support? :) 

Pretty please! :)

Gadgetoid, 2019-11-06:
The LPD8806 is out of scope for this library - which, despite having a mode that (ab)uses SPI, isn't at all SPI based - and would be better served with a regular SPI driver that doesn't require all the magic and sourcery that rpi_ws281x needs to achieve the precise timings of ws281x and compatibles.

Broadly the LPD8806 is pretty much stock SPI, is addressable and can be driven (albeit not as fast as direct memory mapped register bashing) with a handful of lines of pretty straight-forward userspace code. While a unified library for driving pixels would be super nice, unfortunately that's not what rpi_ws281x aims to be.

I can't seem to find a good C/C++ driver for the LPD8806 strips- other than those intended for the Arduino- were you angling for one? 😁 I think the lack of documentation and need for reverse engineering is a large part of this. They don't seem to be very friendly LED drivers, despite being SPI-based.

Snille, 2019-11-06:
Ah, I see. Ok. Thank you for explaining.

"These are not the LED I'm looking for", "I'll move along..." :)

Jokes aside, I came here because the guy who wrote the [RGB Status](https://github.com/EricHigdon/OctoPrint-RGB_status) Plugin for Octoprint uses your library to control the LEDs... :)

closed 2019-11-06 by Snille