jgarff/rpi_ws281x #374: Can’t open /dev/mem: Permission denied

Flimsy-Fox, 2019-10-21:
Hello!

I’m sure this problem has been posted before, but I’m sure I have a unique implementation of this software. I have my code posted below, and whenever I run it, it gives the error above. Only pin  10 works. The other two (pin 13 and 18) require admin rights. The below code is a snippet of a larger project, which also uses PyAudio. I have tried to run the parent script as root, but it completely breaks PyAudio. I need to run two pixel strips simultaneously. Given what I need, it would seem I need to control two led strips without root access. What are your suggestions for me to help solve this?
`from rpi_ws281x import *
strip1 = Adafruit_NeoPixel(68, 10, 800000, 5, False, 255, channel = 0)
strip2 = Adafruit_NeoPixel(68, 13, 800000, 5, False, 255, channel = 0)

strip1.begin()
strip2.begin()`

Flimsy-Fox, 2019-10-25:
I need to have this issue resolved by October 31st.

penfold42, 2019-10-25:
Need ?

I’m not sure you can do what you want.

I suggest you refactor your code so you have one python script running as root driving the strips and then have a non-root script sending the led data to it over fifo or socket

ecriscuolo, 2019-10-28:
I have this exact same problem on a 3B+, using a pure C implementation and PWM1 on GPIO 13 (pin 33).  Running as root (via sudo) solves the problem. 

The init returns a -5 (mmap failure).  I suspect the permissions on /dev/men are the problem. Perhaps careful use of groups could solve the problem.

Gadgetoid, 2019-11-06:
From experience @penfold42's answer is correct- if you wish to run non-root code while still accessing privileged hardware peripherals you will need to split your application into two parts. I've yet to see anyone set up permissions in a way that avoids this.

An old and application specific example of this in practise can be found here: https://github.com/pimoroni/unicorn-hat/tree/master/library_c/unicornd

Flimsy-Fox, 2019-11-19:
It seems the source of the problem is an incompatibility between Portaudio and ALSA.

Thank you for the advice, I will definitely try running a parent script as root with a child script as user.

Flimsy-Fox, 2020-02-08:
Alright. I think I've gotten all the answers I need; thanks for the help everyone!

closed 2020-02-08 by Flimsy-Fox