jgarff/rpi_ws281x #376: Raspbery Pi 4 Rev 1.2

TheChuckWhite, 2019-11-02:
Hi, just got a new Raspberry Pi 4 and tried to run my old code when i get this error message:
`RuntimeError: ws2811_init failed with code -3 (Hardware revision is not supported)
`

Some other people had this problem with older models before and so I checked my Revision with `cat /proc/cpuinfo` and is says 
`Revision	: c03112`
`Model: Raspberry Pi 4 Model B Rev 1.2`

and compared it with the revision in rpihw.c and there was the problem:
`//Raspberry Pi 4`
`.hwver = 0xC03111,`

After a quick change I thought I am done, but now I'm just too stupid to install the lib into the right directory because the -d flag for `setup.py install` is not working (the error message says to specify the lib directory with -d because the default one seems to be wrong for me).

I've only used pip to install libs, so I really don't know how to do that stuff. Hope it gets updated soon.

b4nter, 2019-11-05:
This code might not be compatible with RPI 4 indeed.
See my issue https://github.com/jgarff/rpi_ws281x/issues/372

Also, what exactly RGB lights do you have?

TheChuckWhite, 2019-11-06:
I have some different panels a ton of strips. All are ws2812b.

Gadgetoid, 2019-11-06:
I don't have my hands on the rev 1.2 Pi 4 to confirm that PR #378 works, but a quick check of the revision number confirms that it's valid for a PCB Revision 2 version of the 4GB Pi 4.

No doubt there will need to be similar editions for both the 1GB and 2GB versions- I'll add these premptively since the revision IDs should be predictable.

Huanbo-Sun, 2019-12-12:
Hi there,
I have implemented in such a way that it works:
1. uninstall rpi_ws281x: sudo pip3 uninstall rpi_ws281x
2. download and extract the zip file from https://github.com/jgarff/rpi_ws281x/tree/master
3. modify and save the rpihw.c files: the line ".hwver = 0xC03111" to ".hwver = 0xC03112"
4. sudo python3 setup.py install
5. run the LED script with sudo and it works for me.
Best

closed 2022-01-25 by Gadgetoid