jgarff/rpi_ws281x #381: Raspberry Pi 4 Model B Rev 1.1 SPI trouble

devildant, 2019-11-20:
hello,
I use a neopixel stick and a raspberry pi 4, and I meet a problem with the SPI (gpio 10).
my script displays all the led in red, but a call on two of the script sends the leds in white (luminosity 255)
no problem with the GPIO 18

SteMMo33, 2019-12-11:
Same problem.
Running ok on PWM on GPIO18 or GPIO21, ok for PCM GPIO21.
On GPIO10 I see a signal but the timings are slower for the strip LEDS.

Gadgetoid, 2020-05-15:
Really late to the party here, but I'm running into exactly the same problem.

Interestingly I seem to be able to fix it by running some output on a GPIO pin (in this case, bitbanging APIA102s), which suggests that (at a totally wild guess) the frequency at which the (faster) Pi 4 is updating the SPI output data is exceeding some kind of limit. I'm in way over my head, though.

In my case I get about 7 LEDs of green into `standtest.py` before it bails out and just locks to (eye burningly painful) solid white.

Gadgetoid, 2020-05-15:
@tvoverbeek do you have a moment to weigh in with a hypothesis here?

Gadgetoid, 2020-05-15:
Okay, I don't know why this works, but increasing the SPI speed via changing this multiplier in `ws2811.c` appears to make the LEDs run stably for longer:

```patch
-    uint32_t speed = ws2811->freq * 3;
+    uint32_t speed = ws2811->freq * 8;
```

Eventually, however, they still break- causing rainbow mode to start flashing obnoxiously.

I would have assumed this would break the timing quite spectacularly, since I thought this worked by specially formulating SPI words with consecutive 1s and 0s to simulate the high/low pulses required by the WS281x family.

There is something I don't quite understand at play here.

Gadgetoid, 2020-05-15:
Oh this is just a classic replay of the variable-SPI-clock-speed issue, detailed here: https://github.com/raspberrypi/linux/issues/3381

The whiteout-after-a-short-period-of-time is just the time it takes for the CPU to go from "busy" to "idle", switching down the clock speed which also switches the SPI clock speed, pushing it out of range for the WS281X protocol.

It's telling how obtuse this is, since it took me this long for the penny to drop. In my code where I was bit-banging some other GPIO pins, I was doing enough to keep the CPU busy continuously.

TLDR: Add the following to `/boot/config.txt`

```
core_freq=500
core_freq_min=500
```

shouldsee, 2020-07-17:
Verified that solution works on RPi4 after a long afternoon of debugging.

> Oh this is just a classic replay of the variable-SPI-clock-speed issue, detailed here: [raspberrypi/linux#3381](https://github.com/raspberrypi/linux/issues/3381)
> 
> The whiteout-after-a-short-period-of-time is just the time it takes for the CPU to go from "busy" to "idle", switching down the clock speed which also switches the SPI clock speed, pushing it out of range for the WS281X protocol.
> ...
> 
> TLDR: Add the following to `/boot/config.txt`
> 
> ```
> core_freq=500
> core_freq_min=500
> ```

Gadgetoid, 2020-07-17:
Had a similar afternoon of complete befuddlement when I ran into the same problem. This is why I love APA102s 😆

shouldsee, 2020-07-19:
> Had a similar afternoon of complete befuddlement when I ran into the same problem. This is why I love APA102s laughing

I am so glad to have read this comment while building a POV WS2812. Do you have to know the maximum achievable FPS with APA102 for 1000pixels?