jgarff/rpi_ws281x #384: Raspberry Pi 4B with Neopixel: Hardware revision is not supported

Huanbo-Sun, 2019-12-12:
I tried to use RSP 4B to change the color of Neopixel ring. The code works in RSP 3B with no error, but doesn't work with RSP 4B with the error: "RuntimeError: ws2811_init failed with code -3 (Hardware revision is not supported)". Can anyone solve this problem?

Huanbo-Sun, 2019-12-12:
Hi there,
I have implemented in such a way that it works:

uninstall rpi_ws281x: sudo pip3 uninstall rpi_ws281x
download and extract the zip file from https://github.com/jgarff/rpi_ws281x/tree/master
modify and save the rpihw.c files: the line ".hwver = 0xC03111" to ".hwver = 0xC03112"
sudo python3 setup.py install
run the LED script with sudo and it works for me.
Best

Peer256, 2019-12-19:
Hey Huanbo-Sun,

thank you for your answer, I tried your fix, but ran into another problem installing the setup.py. 

> Hi there,
> I have implemented in such a way that it works:
>
> uninstall rpi_ws281x: sudo pip3 uninstall rpi_ws281x
> download and extract the zip file from [...]

on your proposed last step I ran into an exception: 
after: 
`sudo python3 setup.py install `
...
/usr/bin/ld: -lws2811 could not be found 
collect2: error: ld returned 1 exit status
error: command 'arm-linux-gnueabihf-gcc' failed with exit status 1

Do you have an idea what I did wrong?

Thanks a lot,

Regards.

sharathjapa, 2020-01-03:
run scons in the source folder and then run 
sudo python3 setup.py install

tdejoigny, 2020-01-14:
I used a RPI4 4Go and I have the same error : "RuntimeError: ws2811_init failed with code -3 (Hardware revision is not supported)". Is there a fix ?

anwarbashir, 2020-01-19:
I have the same RuntimeError and I am using a Raspberry Pi 3 A+, I will track this thread becuase I think that a similar fix will be required?

AndreasBrunner, 2020-11-16:
Hello,

i am getting the same problem:

  File "/usr/local/lib/python3.7/dist-packages/rpi_ws281x/rpi_ws281x.py", line 130, in begin
    raise RuntimeError('ws2811_init failed with code {0} ({1})'.format(resp, str_resp))
RuntimeError: ws2811_init failed with code -3 (Hardware revision is not supported)

i am using a raspberry pi 4 8GB. is there any solution for this meanwhile?

AndreasBrunner, 2020-11-16:
a simple:
sudo pip3 uninstall rpi_ws281x
and
sudo pip install rpi_ws281x
did the trick for me

placheckij, 2021-02-27:
> Hey Huanbo-Sun,
>
> thank you for your answer, I tried your fix, but ran into another problem installing the setup.py.
>
>
> on your proposed last step I ran into an exception:
> after:
> `sudo python3 [...]

run command "scons" before "sudo python3 setup.py install" as in https://stackoverflow.com/questions/28380448/cant-install-rpi-ws281x-error-command-gcc-failed-with-exit-status-1 answered by Mattnv92 - it did the trick for me

markyharris, 2021-03-28:
AndreasBrunner's answer was close to working for me. I performed these steps to get it to work;

- sudo pip3 uninstall rpi_ws281x
- sudo pip install rpi_ws281x
- sudo pip3 install rpi_ws281x

After this, my script ran properly.
Thanks

Gadgetoid, 2022-01-25:
See #483 for information on what to do about unsupported revisions.

closed 2022-01-25 by Gadgetoid

daprezjer, 2024-06-27:
> Hi there, I have implemented in such a way that it works:
>
> uninstall rpi_ws281x: sudo pip3 uninstall rpi_ws281x download and extract the zip file from [...]

@Huanbo-Sun you're getting downvotes because there is no setup.py file that would install. Can you advise on how to install with the latest repo? Note for those looking that there are several suggested branched repos out there for newer versions; they're just not official yet.