jgarff/rpi_ws281x #385: Question:  Open Pixel Control Server + rpi-ws281x mashup,  MAKE + SCONS

mizraith, 2019-12-18:
I must admit to being out of my element.

Goal:  Take the proven fadecandy Open Pixel Control server ([GitHub link](https://github.com/scanlime/fadecandy))  and add a route for the rpi-ws281x.   

The fcserver already has a notion of reaching messages over a socket and passing them to multiple endpoints, so adding in the code to init and control pixels via rpi-ws281x seems fairly straightforward.

Except -- rpi-2ws281x seems to have some more complex compile rules in scons and the fcserver is built using make.     

I guess the question is:  what do I actually need to integrate from the rpi-ws281x to make it a component of the fcserver project?    (Seems like much more than a few .c files).

Thanks for anyone's guidance on this.

Gadgetoid, 2022-01-25:
Did you ever get anywhere with this?

fadecandy has kinda... imploded... in the interim.

mizraith, 2022-01-25:
I did get it working to some level - by not using fade candy at all.    The pixel length limitations forbade longer strands.   

I was able to use the rpi-ws281x project directly and socket directly to that to send out pixels.  I shelved the project over the last year or so, but when I left it, I was definitely able to control longer strands, over wifi, at excellent frame rates.