jgarff/rpi_ws281x #387: LED count needs to be 4 higher than the actual number of LEDs

BowenDragonheart, 2020-01-06:
Hi,

I am trying to build my own Word Clock and am in the final stage of testing the LED strips.

Using the both strandtest.py and test script of the source directory, I run into a strange problem: I need to specifiy four more LEDs than are actually in the LEDstring to have all light up.

I'm using a Raspberry Pi Zero W, combined with a Mean Well power supply (5V, 50W) and the a W2812B LEDstrip

(https://www.banggood.com/1M-5M-WS2812B-5-Pins-RGBW-RGBWW-4-IN-1-LED-Strip-Light-Non-Waterproof-DC5V-p-1211152.html?rmmds=myorder&ID=51619044707&cur_warehouse=CN)

I power the Pi through the USB and not GPIO 2 and 6 at the moment. The LEDstrip and Pi do share a common ground.

Can somebody please help me out?

Thank you!

Best regards,
Bart

BowenDragonheart, 2020-01-06:
I did a few more tests.

If I specify LED count to be 7, than the sixth LED only shows the color green.
For a LED count of 8, the sixth LED seems to have the full RGB again.

Gadgetoid, 2020-01-15:
Don't forget to specify a strip type as demonstrated here - https://github.com/jgarff/rpi_ws281x/blob/master/python/examples/SK6812_white_test.py

Your LEDs have 4 elements, not 3, so they need an extra byte of data per LED.