jgarff/rpi_ws281x #388: Unicorn HAT HD Runtime Error: ws2811_init failed with code -3 (Hardware revision is not supported)

mlcmds, 2020-01-15:
Runtime Error: ws2811_init failed with code -3 (Hardware revision is not supported)

Running on Raspberry PI 4
Using python library rpi-ws281x  Version 4.2.2
Python version 2.7 
Getting error
Runtime Error: ws2811_init failed with code -3 (Hardware revision is not supported)
Processor revision is c03112

cat /proc/cpuinfo | grep Revision  gives 
           Revision   : c03112

Kindly let me know how to avoid this error using this python library.
Best Regards
Balachander

Gadgetoid, 2020-01-15:
It looks like I never built/pushed rpi-ws281x 4.2.3 which has support for a handful of Pi 4 models. Sorting that now.

mlcmds, 2020-01-15:
Thanks.

Looks like rpi-ws281x 4.2.3 is available. It works

Regards
Balachander

closed 2020-06-07 by Gadgetoid