jgarff/rpi_ws281x #390: ws2811_init failed: Out of memory

robertf26, 2020-01-17:
Hi,
I've been driving a WS2812B strip (300LEDs) successfully with a Raspi 1B (512MB RAM) Raspbian Stretch without GUI for a few years with this software.
Recently I added new software to my Pi and noticed that I can send LED signals initially, but after long uptimes of the Pi (between a few hours and a few days) I get: `ws2811_init failed: Out of memory`
In such a situation `free -m` looks like this:
``` 
              total        used        free      shared  buff/cache   available
Mem:            480         340          40           5          99          87
Swap:            99          18          81
```
Only a reboot fixes this.
Now I have been trying to increase the RAM available to Raspbian by reducing the VRAM to a minimum of 16MB since I have no GUI. But somehow it seems that it made things worse.
Now I skimmed the source code and saw a reference to VideoCore and was wondering wether this program actually uses VRAM and that reducing such memory was counterproductive. It would be great if someone could answer as I barely understand C code.
Sadly I cannot upgrade my Pi as I have a kernel module which will only work with single core CPUs.

Gadgetoid, 2020-01-17:
I'm fairly sure that rpi_ws281x doesn't use VRAM at all. The out-of-memory exception is probably due to a memory leak somewhere in either this library or some other related code.

Judging by the error being on `ws2811_init` I'm guessing you're running a script periodically (crontab?)? Since `init` should only be called once.

jgarff, 2020-01-17:
Hey guys,

There is actually a small amount of VRAM being used by the library.  We
need to use it because the memory where we generate the output bit stream,
needs to be cache coherent with the hardware DMA/PWM/PCM/etc..  So you do
need to make sure you have some, but probably not much, video memory
available.

I completely agree with Philip though that the library probably isn't the
cause of the memory leak, though it may be something on top of it.  You
should be able to use some system tools to determine which task is
consuming all of the memory.  As for the video memory, it's confined to
whatever size you setup during boot.

Hope that helps,

Jeremy

robertf26, 2020-01-20:
Thank you for the replies! I will further investigate where the memory leak comes from when I have more time. Meanwhile a nightly cronjob to reboot the Pi does the job.

dustyyeti, 2020-03-21:
Hi, I have a similar issue. 5 minute increment cronjob changes output to neopixels on GPIO. 
The cronjob invokes a shell script that then calls a circuit python script to change the lights.

In the log files I find this occurs after two days of running:
`RuntimeError: ws2811_init failed with code -2 (Out of memory)`

@robertf26 did you come up with any leads?

Xartrick, 2020-03-23:
> Hi, I have a similar issue. 5 minute increment cronjob changes output to neopixels on GPIO.
> The cronjob invokes a shell script that then calls a circuit python script to change the lights.
>
> In the [...]

Do you call `ws2811_fini` at then end of your script?

dustyyeti, 2020-04-06:
> Do you call `ws2811_fini` at then end of your script?

Sorry, not used to catching notifications on Github.

Thanks for your comment.

I'm using the Adafuit CircuitPython NeoPixel libraries. 
https://github.com/adafruit/Adafruit_CircuitPython_NeoPixel

My python script imports: `neopixel` and `RPi.GPIO`

I don't issue any commands directly to ws2811, and I don't see any code for ws2811 in the repo.

I looked for examples of `ws2811_fini` and found some python that does in fact initialize and finalize like you say, but it also has an instantiated object to call it. Like:

```
resp = ws.ws2811_init(leds)
.
[later]
.
ws.ws2811_fini(leds)
```

I did clone this repo [ws281x] but don't remember what step of my installation asked for that, or how I use it. My assumption is that the adafruit library must require it? I'm out of my depth here.

Eventually I put a daily reboot order in my crontab as a hacky work around. The reboot happens in time to clear the memory leak.

What would you suggest?

lynolamero, 2022-12-13:
> Thank you for the replies! I will further investigate where the memory leak comes from when I have more time. Meanwhile a nightly cronjob to reboot the Pi does the job.

Will you share with us the code to do this?