jgarff/rpi_ws281x #391: GPIO pins acts "together" behaviour

szafranski, 2020-03-03:
I found very weird issue/behaviour. When LED pin is changed to "12", both pins GPIO 12 and GPIO 18 are outputting the same signal. Looks like they are somehow "tied" together. Same behaviour occurs when there is "18" provided in the config file - GPIO 12 and GPIO 18 output the signal. So looks like when pin GPIO 18 is used, GPIO 12 cannot be used for other purposes. Few questions:
1. Is this a known issue? 
2. Are there any other pin "pairs" that cannot be used together?
3. Can MOSI on SPI1 bus on Raspberry Pi 3 B+ be used as data pin?

Gadgetoid, 2020-03-10:
Pins BCM12 and BCM18 are indeed both PWM0, so if they're muxed to PWM mode then they will always have the same output. (mode Alt5 is PWM0 on BCM18 and mode Alt0 is PWM0 on BCM12)

You should be able to switch either BCM12 or BCM18 (depending on which pin you happen to be using to drive the LEDs) into regular GPIO mode (or an available alt mode) in order to use it elsewhere.

Both pins being switched into PWM mode when you're only attempting to use one *might* be a bug.

szafranski, 2020-03-10:
They are acting together in pwm mode. Can they be untangled?

szafranski, 2020-04-04:
I just got what you wrote. Thanks. So until that another pin is changed (in some another program etc) it will always act "together"?

jgarff, 2020-04-04:
I just looked at the lower layer C code.  It should only switch the pin
you've selected, not both.  I can't speak for higher layers though
unfortunately.

It would also appear that these pins aren't changed back to previous
settings on program exit, so it's possible that if you're playing with
different ports, and haven't rebooted in between, this could also explain
the behavior.

szafranski, 2020-04-04:
I was switching pins back and forth without reboot so that will explain
that situation probably.

closed 2020-06-07 by Gadgetoid