jgarff/rpi_ws281x #393: Scons fails: 'SyntaxError: leading zeros in decimal integer literals are not permitted'

bepaald, 2020-03-09:
Build fails when executing lin 87 of SConscript: https://github.com/jgarff/rpi_ws281x/blob/60e7fc4d90640ecf875bd555d44d5381ec6843c1/SConscript#L87

```
scons: Reading SConscript files ...
  File "/home/bebr/.cache/pikaur/build/rpi_ws281x-git/src/rpi_ws281x/SConscript", line 87

    Chmod("$TARGET", 0755)

                        ^

SyntaxError: leading zeros in decimal integer literals are not permitted; use an 0o prefix for octal integers
```

The (current) [scons user manual](https://scons.org/doc/3.1.2/HTML/scons-user/ch12s06.html) says to use `Chmod("$TARGET", 0o755)` instead.

Thanks

Xartrick, 2020-03-09:
Hi @bepaald.

The only similar bug report I found is here : https://aur.archlinux.org/packages/rpi_ws281x-git#comment-718773

Xartrick, 2020-03-12:
Issue can be closed, PR merged!

closed 2020-03-13 by Gadgetoid