jgarff/rpi_ws281x #396: Using GPIO 18 requires sudo permissions

szafranski, 2020-04-01:
Without sudo I get on Rapsberry Pi 3b+ an error like this:

Can't open /dev/mem: Operation not permitted. 

Any safe solutions?

Thanks!

szafranski, 2020-04-04:
Using GPIO 10 (SPI) works without sudo. What can be an issue with GPIO 18?

penfold42, 2020-04-07:
It uses PWM and there’s no way to access the PWM hardware as non-root

szafranski, 2020-04-08:
So pwm neede sudo and spi doesn’t right? 
Is there any way to change those permissions than?

penfold42, 2020-04-08:
Correct

And unless/until the kernel exposes a PWM driver to user space, no you need root

szafranski, 2020-04-08:
But if I just use shell script with simple:

gpio -g mode 12 pwm
gpio -g pwm 12 1024

It runs without sudo. How is that possible?

penfold42, 2020-04-08:
That’s only setting the pin to a fixed pwm value.

Driving 1 wire leds hits the hardware directly to rapidly send the pulse codes to meet the tight timing requirements

szafranski, 2020-04-09:
So is there some threshold frequency value above which U have to use sudo? Of course I can make some for loops etc and change pwm in a script without sudo so I am curious how it works.

jgarff, 2020-04-09:
That will likely not work for the following reasons:

- The frequency may be too low using bitbanging on a GPIO.
- The Linux scheduler will swap out the task thereby causing too much
jitter in the output waveform.

This is why there aren't already GPIO based bitbanging solutions, and also
why it requires root.  You need hardware assist to make it work properly.
You either need to use Pi built in hardware (PWM + DMA) or another RTOS
type uC like a arduino via USB.

szafranski, 2020-04-09:
I’m am designing a PCB for quite large group of people. I would like to use
pin 18 due to phisical placement cause everything else is „nearby” like
uart and i2c. Is there ANY walaround like changing permissions or sth?

Thanks for all responses!

jgarff, 2020-04-09:
Unfortunately, not without sacrificing security.  Changing the user/group
on /dev/mem for example is a possibility, but is not recommended.

szafranski, 2020-04-09:
I would like to keep dev/mem as it is. Any possible alternative? Like
changing to some part of it?

jgarff, 2020-04-09:
Sorry, no.

szafranski, 2020-04-09:
Thank you, that library is awesome anyways. What would be the closest pin
that can be used on Raspberry pi 3b+ than?

Cheers

szafranski, 2020-04-09:
Don’t have to be pwm pin. Just closest to uart and i2c.

fmatray, 2020-04-13:
Hello,
I would like to create a plugin for mopidy to add leds cause we all love colorful things :-) 
The only pin not used by the pirate audio hat is the 12 but I don't want to run mopidy as root.