jgarff/rpi_ws281x #397: Random flickering in some-cases

JPZV, 2020-04-26:
I have a 200 Leds WS2812b strip powered by a dedicated 12v PSU (I don't remember the output current, but I know that was enough to power the strip) that I use for ambient lighting my room.
I built this a long time ago (maybe one year ago, or even more) and it was working perfectly. Then I updated the setup to include addressable leds feature and I made a [Pristmatik](https://github.com/psieg/Lightpack)'s plugin to sync the leds with my screen, again it worked perfectly, but I didn't use it a lot.

One/two months ago, I wanted to use the Pristmatik feature but I did noticed that sometimes some Leds just do a white blink randomly. That flickering wasn't even constant, but it occurs in a random time in some random leds. I did a lot of troubleshooting, from looking my plugin's output/input (I thought that it was sending an wrong package) to even buying a level shifter (I didn't used one before as I didn't know that it was "necessary" because the lights has been working from the beginning).

So, from then, in my free time (that I don't have a lot), I tried this:

0) Updating rpi_ws281x from 1.x.x (I think) to the latest update (2.4.3) -> Nothing new
1) Using the neopixel_rpi_simpletest.py -> It works but I get random blinking too
2) Changing the old data wire by a new ones, even I did changed the method to soldering the Pin connector. Also, before I was using a generic CAT5 (maybe E, I don't know) not shielded ethernet cable, now I'm using a CAT5E FTP 24AWG ethernet cable -> Nothing new
3) Connecting the RPi and Strip Ground to the another PSU's Ground output -> Nothing new
4) Connecting the RPi to the same PSU's ground Output as the Strip but with another wire (so each one has its own dedicated wire) -> Nothing new
5) Buying a Level Shifter to convert the 3.3(ich)v to 5v data signal -> Nothing new
6) Using another GPIO port for data -> Nothing new
7) Stopping MariaDB and Apache2 Services -> Nothing new
8) Use os.nice(-20) -> Nothing new

Also I did some test:

1) Use another script to change all the lights at once only 1 time -> Works perfectly.
2) Use the same script from 1) to change all the lights multiple times -> I think it works perfectly.
3) Lowing the refresh rate from neopixel_rpi_simpletest.py -> It blinks but "less"
4) Lowing the LED Count from 200 to 50/30/20 in neopixel_rpi_simpletest.py -> It stills blinking

Also, I have a AC Splitter connected to the same wall outlet as the RPi, and I use that splitter to connect an Soldering Iron, and sometimes somes leds turns on once in a random position with a random color with a random brightness for each one. So I think that there should be grounding problem or even an interference, so I wanted to connect an resistence between the Data and the 5V/3.3V/Gnd (one of them of course), but I don't know how to determinate that or even how to determinate how much Ohm should I use. I tried searching in Google but I didn't found anything.
*Note that both the PSU and the RPi doesn't share any outlet connector.

I will upload a (visual) Scheme later, but this is my wiring:

a- RPi 3.3v Output    -> Level Shifter's LV (Low-Volt Input)
b- RPi 5v Output      -> Level Shifter's HV (High-Volt Input)
c- RPi D21 (Pin 40) -> Level Shifter's L1
d- Strip Data             -> Level Shifter's H1
e- RPi Ground  ->  Level Shifter's LG (Low-volt Ground)  ->  Level Shifter's HG (High-volt Ground)  ->  Strip ground  -> PSU Ground (This is the current setup, but before I didn't did that loop setup, and stills doesn't work in that way).

I'm using a Raspberry Pi 3b with Python 3.7.3 and Raspbian Buster (with everything updated, even the PIP packages).

Gadgetoid, 2020-06-07:
Forgive me if I missed it- but which version of the Raspberry Pi are you using?

My only hunch is that you're seeing CPU frequency scaling periodically throwing your LED signal out of range. I know this affects SPI, but I'm not sure about PCM/PWM modes.

JPZV, 2020-06-07:
I'm using a Raspberry Pi 3 B (launched in 2016) with Raspbian 10 (Buster).

And about your hunch, it could be right, as when it was working fine like one year ago, I had the RPi floating (almost literally) in the wall, but now it's inclosed in a case for the Official Touch Screen (the case doesn't have enough room. In fact it doesn't has support for a fan), so now it's running a little hot (like 58°C Max and maybe 50°C AVG). The problem here is that I get flashing light even with 50°C and just using 26% of a single core (all the other cores are almost 0%). So I don't know if I'm getting any type of CPU throttling.

Gadgetoid, 2020-07-23:
You should try adding in-line current-limiting resistors to your setup, from your step-by-step it doesn't look like you've attempted this.

This is probably something for the documentation, since it seems to be cropping up from time to time as a problem - to the extent we've populated resistors on our work-in-progress LED driver HAT.

I haven't had this issue myself to take a scope to it, but my very rough guess would be that a lack of resistor allows for a high inrush current that causes "ringing" sufficient enough for the LEDs to interpret it as a signal.

I'll find out what value we used, in case it helps you.