jgarff/rpi_ws281x #398: use fill

qiwalker, 2020-05-05:
hi, need some help
i have 20 ws2812 RGB
i want to light up all the LED to green color 
i try to use   strip.fill((0, 255, 0)) nothing happen :(

Xartrick, 2020-05-05:
Hey there!

We need more information about your issue if you want us to help you. I will assume you're using the Python bindings.

- Are you sure that the `fill` method exists? I don't see any reference to it in the code.
- Are you able to light up a single LED using `setPixelColor` and `show`?

Also, can you give use details about your wiring and also a piece of code you use?

Thank you.

penfold42, 2020-05-06:
Are you calling strip.show()

Do the examples work for you ?

up2smoke, 2021-01-29:
+1
how do you put all LEDs of a strip on the same color?

Gadgetoid, 2021-12-03:
This looks like a Python question. If this is still a problem you can raise an issue over on the Python bindings - https://github.com/rpi-ws281x/rpi-ws281x-python

closed 2021-12-03 by Gadgetoid