jgarff/rpi_ws281x #418: "test" program does not support Channel 1

combs, 2020-10-15:
I believe that the `test` program compiled from `main.c` doesn't support the use of Channel 1, due to the following hardcoded values:

```
ws2811_t ledstring =
{
    .freq = TARGET_FREQ,
    .dmanum = DMA,
    .channel =
    {
        [0] =
        {
            .gpionum = GPIO_PIN,
            .count = LED_COUNT,
            .invert = 0,
            .brightness = 255,
            .strip_type = STRIP_TYPE,
        },
        [1] =
        {
            .gpionum = 0,
            .count = 0,
            .invert = 0,
            .brightness = 0,
        },
    },
};

```

Some configurations (e.g. GPIO 13, PWM1) require the use of Channel 1, and it outputs confusing errors:

```
$ ./test -g 13
Gpio 13 is illegal for LED channel 0
ws2811_init failed: Selected GPIO not possible
$ ./test --help 2>&1 |  grep -i channel
-d (--dma)     - dma channel to use (default 10)
$ # can set DMA channel but not LED channel
```

possible fixes:

- Extend `test` to allow selection of LED channel 1 with a new argv. Populate [1] instead of [0] with user-supplied values
- Add a note to the README explaining this to avoid heart attacks in others who built custom hardware with GPIO 13 for ws281x leds and are testing out the configuration for the first time ;)

thanks y'all--this is a fantastic project and I hope this is helpful.

gdhgdhgdh, 2020-10-28:
I'm in exactly the same boat - did you find any solution (even if very hacky!) ? I want to get the `test` working and then hack it into the end of another application.

combs, 2020-10-28:
Yes--my hacked-up Channel 1 only main.c is attached:

combs, 2020-10-28:
[main-channel-1.zip](https://github.com/jgarff/rpi_ws281x/files/5452502/main-channel-1.zip)

gdhgdhgdh, 2020-10-28:
> Yes--my hacked-up Channel 1 only main.c is attached: https://github.com/jgarff/rpi_ws281x/files/5452505/main-channel-1.zip

Thank you :) I deleted my original comment because I figured it out - however I appreciate you taking the time to reply and help <3