jgarff/rpi_ws281x #421: GUI, Home Assistant or other uses?

charlesomer, 2020-10-27:
Hi,

I’ve been trying to find out if there’s a well maintained GUI, or Home Assistant plugin etc. which uses this project (or similar method to control LEDs on a Pi) but I can’t seem to find anything.

Before starting anything myself I just wanted to make sure I haven’t missed anything obvious that you’re aware of?

Thank you
Charles

Gadgetoid, 2020-11-04:
There are none that I'm personally aware of... unless you include the little paint application I wrote for our Unicorn HAT many, many moons ago.

I have a work-in-progress to turn our Plasma lighting library into a general purpose smart-LED library that will drive APA102/WS281X with various configurations and combine them into a single strip for animations and effects - https://github.com/pimoroni/plasma/pull/7

This includes a lighting daemon that can, among other things, sequence the LEDs from a PNG image with cols for LEDs and rows for animation steps. It might have some feature crossover with what you want to build.

cp2004, 2020-11-26:
For OctoPrint, I created a plugin - [WS281x LED Status](https://plugins.octoprint.org/ws281x_led_status) - if you want to check it out. Uses the SPI side of the library to communicate with the LED strip.