jgarff/rpi_ws281x #424: undefined reference to `pow' as of #409

TwinProduction, 2020-11-10:
As of https://github.com/jgarff/rpi_ws281x/pull/409, I am getting the following error:

```
# github.com/rpi-ws281x/rpi-ws281x-go
/usr/bin/ld: //usr/local/lib/libws2811.a(ws2811.o): in function `ws2811_set_custom_gamma_factor':
/home/pi/workspace/rpi_ws281x/ws2811.c:1298: undefined reference to `pow'
collect2: error: ld returned 1 exit status
```

I'm aware that I'm using this library through another repository, but I believe that the root cause lies within this repository (or more precisely, since the aforementioned PR (#409)).

I've created an issue on that library as well for traceability: https://github.com/rpi-ws281x/rpi-ws281x-go/issues/12

Xartrick, 2020-11-15:
Seems that `libm` (lib math) isn't correctly linked to your project.

TwinProduction, 2020-11-15:
You were right

Thank you @Xartrick

closed 2020-11-15 by TwinProduction

Silberlachs, 2020-12-16:
Well, how DID you solve it?
Link against libm HOW?
All I can do is "go get [ this project ]"

Silberlachs, 2020-12-16:
I have exactly this issue. would be nice to have a tipp of actually linking it against clibm instead of just pointing out the obvious.
I'm just calling "go get [link_to_repository]" , i can't see any way of passing linker flags here

DeepLightLabs, 2021-01-05:
Could someone elaborate on the solution of linking `libm`?  I am encountering the same error here.  Not sure how to reference this lib to correctly link.  @Xartrick

KC5-BP, 2021-12-08:
Hi there ^^

Link libm or the math lib. to the cmd as follow :
$ gcc main.c -lws2811 -lm -o <yourOutputName>
Also find the ref. here 
https://stackoverflow.com/questions/5419366/why-do-i-have-to-explicitly-link-with-libm

nmerchant, 2021-12-09:
> Hi there ^^
> 
> Link libm or the math lib. to the cmd as follow : $ gcc main.c -lws2811 -lm -o Also find the ref. here https://stackoverflow.com/questions/5419366/why-do-i-have-to-explicitly-link-with-libm

Thanks so much for the reply!  Running that command returns:
```
gcc: error: missing filename after ‘-o’
```
Any idea what should go after -o?