jgarff/rpi_ws281x #426: Dart binding for rpi_ws281x

darmar-lt, 2020-11-16:
Recently I developed Dart binding for rpi_ws281x:
  https://github.com/darmar-lt/rpi-ws281x-dart

I hope, somebody will find it useful.

Gadgetoid, 2021-12-03:
Feel free to raise a PR to add this to the list in the README. We also have a GitHub Org for collecting rpi_ws2812 bindings if you're interested - https://github.com/rpi-ws281x/

darmar-lt, 2021-12-30:
Hello Gadgetoid,
I think, it would be nice to have rpi-ws281x-dart as one of the bindings on https://github.com/rpi-ws281x/ 
Can you add it there?

Gadgetoid, 2022-01-04:
I believe that's something @penfold42 has to facilitate. For some reason I thought I'd set up that org, but apparently not!

darmar-lt, 2022-01-08:
Thank you Gadgetoid. Unfortunately @penfold42 doesn't write his email. Therefore I have no way to contact him/her.

penfold42, 2022-01-08:
Done - I think !

darmar-lt, 2022-01-08:
@penfold42: Thank you!

Gadgetoid, 2022-01-25:
> him/her

Singular "them" for unknown third person :smiley: 

I think this is resolved, now. Glad to have you aboard!

closed 2022-01-25 by Gadgetoid