jgarff/rpi_ws281x #428: Too big pause between frames

Gregor14, 2020-11-30:
Hi everybody,
this is my first post on Github, so please be gently :)
Probably I met small error in code, where program try to establish proper pause between frames. Now in code is as follow:
https://github.com/jgarff/rpi_ws281x/blob/3a20f1bdd740a144de180a07d8d3317920459d2d/ws2811.c#L1248-L1268

In this way previous_timestamp is evaluated after frame is sent. So, pause between frames will consist not only LED_RESET_WAIT_TIME but also time of whole frame.
In my opinion line "previous_timestamp = get_microsecond_timestamp();" should be just before "if(driver_mode != SPI)". In this way pause between frames will become corrected (shorted). 
Solution was implemented by me few months ago, works well.

McAleerElk, 2021-05-15:
Hey. Good observation. Might come in handy. How many LED's were you controlling at the time? 

With the libraries I tried, I experienced delay between frames, purely due to the amount of calculations needed per frame. Eventually, I am looking to run 2450 which is why Python seems to operate too slow. 

I think I'm in deep water here, but I haven't given up hope. This library here is the next one I'm going to try out.

artcontr, 2021-05-24:
Hi all, I'm also looking to run a large number of leds (1440 rgbw) at 25fps so if anyone has any idea for me please :-)

Gregor14, 2021-05-24:
> Hey. Good observation. Might come in handy. How many LED's were you controlling at the time?

My case was a little bit differ. I'm using 8 quite short strings of LED's (up to 40 LEDs per each). All of them are connected through demux (74HC238). My goal was to drive them all (one after one) as fast I can. That's why, this enourmous time gap between datas was for me a problem. With correction what I show above, whole system works well since 6 months.

jgarff, 2021-05-24:
The time it takes per channel using this library depends mostly on the
frequency of the WS2812 signal rate (800Khz).  So, you can calculate
roughly as follows:

Bits per pixel (32 for rgbw)
Number of pixels (1440)

So typically something like this applies:

800000 / 32 / 1440 = ~17 FPS.

artcontr, 2021-05-24:
Hi.
So it means I could use 2 channels with 720 leds on each channel and reach
800000/32/720=34 fps (so my 25fps target easily) ?
What mode should i use ? Can anyone point me in the right direction or
maybe provide paid advice ? 🙂
Best regards.