jgarff/rpi_ws281x #431: Raspberry Pi 4 Model B Rev 1.4 - Hardware Revision Not Supported

frid1993, 2020-12-11:
Software faults out at the 'init' step. Can anyone advise?

RussellTaylor83, 2020-12-26:
I'm in the same boat, have referenced this issue from here

https://github.com/Freenove/Freenove_Robot_Dog_Kit_for_Raspberry_Pi/issues/6

RussellTaylor83, 2020-12-26:
```bash
Traceback (most recent call last):
  File "test.py", line 3, in <module>
    led=Led()
  File "/home/pi/git/Freenove_Robot_Dog_Kit_for_Raspberry_Pi/Code/Server/Led.py", line 22, in __init__
    self.strip.begin()
  File "/usr/local/lib/python3.7/dist-packages/rpi_ws281x/rpi_ws281x.py", line 131, in begin
    raise RuntimeError('ws2811_init failed with code {0} ({1})'.format(resp, str_resp))
RuntimeError: ws2811_init failed with code -3 (Hardware revision is not supported)
Segmentation fault
```

rip3rs, 2021-10-21:
any solution found? getting the same issues

Gadgetoid, 2022-01-25:
See the very helpful guide here: https://github.com/jgarff/rpi_ws281x/issues/483 and - if you can - make a pull-request for your newly discovered revision ID.

We get no advance warning of new Pi models, so no idea what to add until someone finds them out in the field.

Hopefully this bugbear is soon going away in favour of https://github.com/jgarff/rpi_ws281x/pull/183

closed 2022-01-25 by Gadgetoid