jgarff/rpi_ws281x #433: WS2815 needs 12V power supply

baqwas, 2020-12-19:
Hello!

Just a minor documentation update please.

The statement "WS281X LEDs are generally driven at 5V" needs a very minor correction. **WS2815** requires a 12V power supply. Apologies for posting this note as an issue but I feel that those embarking on exercises with **WS2815** need to recognize the need since after WS2812B the suppliers have the text (_viz._ **WS2815**) working model in the long description of the SKU but the buyer may not be alert to the power supply requirement for such a model. Thanks.

Kind regards.

**P.S.** Thanks also for the efforts in porting the libraries to the RPi world.

jawsware, 2021-07-05:
Does this mean you got the WS2815 working?  How did you wire the backup data?

baqwas, 2021-08-08:
Corresponding wire AWG is important too.

closed 2021-08-08 by baqwas