jgarff/rpi_ws281x #439: Scrolling text with WS2812b matrix (32x8)

markusweibofner, 2021-01-05:
Hi there, 
are there (Python-based) algorithms based on this repository, which translate text into the pixels required for the WS2812b? I could find a lot for Arduino, but unfortunately nothing for the Raspberry.
I've found a lot of code, but mostly for different LED modules.
Any help appreciated!
Thank you!

kevinIvan, 2021-10-07:
Hi there,
take a look on that nice job: https://github.com/jbentham/rpi

Gadgetoid, 2021-12-03:
Since rpi_ws281x is linear and you want to use 2D coordinates, assuming you have wired a striped matrix somehow it's just:

```python
index = y * WIDTH + x
```

If you have a snake-style matrix (winding the pixels back and forth to form a rectangle) it's easier just to use a lookup table and not lose any sleep over it. eg:

https://github.com/pimoroni/unicorn-hat/blob/657deffc385895d43a81828c30355e5051cce6c7/library/UnicornHat/unicornhat.py#L76-L85

```python
HAT = [
    [7 , 6 , 5 , 4 , 3 , 2 , 1 , 0 ],
    [8 , 9 , 10, 11, 12, 13, 14, 15],
    [23, 22, 21, 20, 19, 18, 17, 16],
    [24, 25, 26, 27, 28, 29, 30, 31],
    [39, 38, 37, 36, 35, 34, 33, 32],
    [40, 41, 42, 43, 44, 45, 46, 47],
    [55, 54, 53, 52, 51, 50, 49, 48],
    [56, 57, 58, 59, 60, 61, 62, 63]
]
```

This is just a 2D list you can index into with X/Y coordinates to return an index:

```python
index = HAT[y][x]
```

It's not much more complicated than that. Feel free to pillage the Unicorn HAT library and examples since it's pretty much exactly what you're looking for (it's based on rpi-ws281x-python) only 8x8 instead of 32x8.

closed 2021-12-03 by Gadgetoid