jgarff/rpi_ws281x #440: WS2811 (12V / 400KHz) + Raspberry Pi ZERO W -> Random blinks + SPI not work.

piotrGTX, 2021-01-17:
RaspberryPi Zero W (Rev. 1.1)
WS2811 (12V 400KHz)
Logic level converter (3.3V > 5V)
GND of Raspberry and power supply is connected

On PWM & PCM when I run "test" script. 
Color wave is correct but red background blinks.
When I try my own script I also saw random colors & blinks (but not all points)

On SPI colors are totally random. I can't see color wave.

I tried to change /boot/config.txt by adding:
core_freq=250
core_freq_min=250 

Then the SPI starts work but works worst than PWM & SPI.... (More blinks and random colors)

Maybe something is not ok with clock & 400KHz on RaspberryPi Zero ?

piotrGTX, 2021-01-17:
When I run "examples/strandtest.py" on PCM

- Color wipe animations. -> OK
- Theater chase animations. -> OK
- Rainbow animations. -> OK (when there are few colors). NO_OK (when there are a lot of colors)

Xartrick, 2021-01-17:
I had almost the same issue.

I used a 16V 1000µF capacitor between +12V and ground. This helps avoiding power spikes which could damage leds. Also, a 300 Ω resistor to avoid electrical noise.

You should wire them close to the ledstrip.

piotrGTX, 2021-01-17:
> I had almost the same issue.
> 
> I used a 16V 1000µF capacitor between +12V and ground. This helps avoiding power spikes which could damage leds. Also, a 300 Ω resistor to avoid electrical noise.
> 
> You should wire them close to the ledstrip.

I do not have 300Ω resistor. So I tried to connect two 500Ω resistors in parallel, but it works worse 😢 (more flashes). With 500Ω resistor does not work at all.

Xartrick, 2021-01-17:
I'm no expert in electronic, but I think 500Ω might be too much and your ledstrip won't correctly pick up low and high signal.
I'm using exactly a 230Ω.

Did you check you power usage? See if the power supply is following up.

piotrGTX, 2021-01-17:
> I'm no expert in electronic, but I think 500Ω might be too much and your ledstrip won't correctly pick up low and high signal.
> I'm using exactly a 230Ω.
> 
> Did you check you power usage? See if the power supply is following up.

Woow i try with these resistors and SPI with and it works! The last led is the wrong color but the others work perfectly :)

piotrGTX, 2021-01-17:
Now SPI works without changing `core_freq` 👍

closed 2021-01-17 by piotrGTX