jgarff/rpi_ws281x #446: wrong led colors, but not always

tysonnorris, 2021-02-14:
I wrote a simple test script to blink my ws2812 LEDs, and I mostly get the wrong colors, but not always. 

Example script:
```import time

from rpi_ws281x import PixelStrip, Color, ws

# LED strip configuration:
LED_COUNT = 12        # Number of LED pixels.
LED_PIN = 12          # GPIO pin connected to the pixels (must support PWM!).
LED_FREQ_HZ = 800000  # LED signal frequency in hertz (usually 800khz)
LED_DMA = 10          # DMA channel to use for generating signal (try 10)
LED_BRIGHTNESS = 2  # Set to 0 for darkest and 255 for brightest
# True to invert the signal (when using NPN transistor level shift)
LED_INVERT = False
LED_CHANNEL=0
LED_TYPE=ws.WS2812_STRIP
LED_GAMMA=None 

# Main program logic follows:
if __name__ == '__main__':
    strip = PixelStrip(
        LED_COUNT, LED_PIN, LED_FREQ_HZ, LED_DMA, LED_INVERT, LED_BRIGHTNESS, LED_CHANNEL, LED_TYPE, LED_GAMMA)

    # Intialize the library (must be called once before other functions).
    strip.begin()

    # reset the pixels
    for i in range(0, strip.numPixels(), 1):
        strip.setPixelColor(i, Color(0, 0, 0))
    strip.show()

    time.sleep(1)
    
    strip.show()

    R=Color(255, 0, 0)
    G=Color(0,255, 0)
    B=Color(0,0,255)

    while True:
      for i in range(0, 10, 4):                 
        strip.setPixelColor(i, R)
        strip.setPixelColor(i+1, G)
        strip.setPixelColor(i+2, B)
      strip.show()     
      time.sleep(1)
      for i in range(0, strip.numPixels(), 1):
        strip.setPixelColor(i, Color(0, 0, 0))
      strip.show()
      time.sleep(1)
```
Expected: 
- every 1 second, every 4 pixels should be R, G, B, dim, stay on for 1s, then off 1s
- the blink is correct, but the color is most often incorrect - but not always; see attached video - you can see that there are periodically moments where the expected R + G + B colors are shown, but mostly never twice in a row? At the end of the video, there are 2 in a row with proper colors, but otherwise, only a few times where proper colors show.

It seems like the gamma is wrong, sometimes, but not always.  I'm not sure what might cause this?

I have tried setting brightness max etc with no difference. 

Is there some way to completely disable gamma, in case that is causing issues?

https://user-images.githubusercontent.com/1638396/107887438-fcd93d80-6eba-11eb-8dd4-aae65b1a538a.mov

tysonnorris, 2021-02-14:
A clue - using SPI, GPIO10, works consistently and correctly. I must have some interference on PWM GPIO12. I would prefer not use SPI for LEDs for unrelated reasons, but if there was a way to use CE1 to it may be feasible, but I didn't read any indication that this library supports chip select signals?

frederico-klein, 2021-10-08:
Did you manage to find a fix for this? 

Is it changing the pin? from 12 to 10?