jgarff/rpi_ws281x #447: SPI LEDs Stuck White

robert-nash, 2021-02-18:
Hello,

I've been having some awful difficulty trying to get some LED strips to work using SPI. Everything works fine using PWM (GPIO 18) but if I use SPI (GPIO 10) the LEDs will only turn white and then won't turn off.

Only LEDs that are meant to turn on turn on. I.e. if I run the test script to run on 8 LEDs, 8 will turn on just they are white at what appears to be full brightness. They then won't turn off without removing the power to the LEDs.

Unfortunately I am stuck using GPIO 10 as it is the only suitable pin I have remaining because I need to use the other pins for driving a separate LED matrix (the problem persists even when the matrix is not involved).

I am using a raspberry pi 3 B+ and I have added core_freq=250 to config.txt. Other issues seem to ask about level shifters so for what it's worth I am using a TXS0108E.

I know that this isn't a whole load of information to go on but  if someone could point me in the right direction that would be much appreciated.

GAEGAEGAE, 2021-06-01:
same proble. somebody help me

haenno, 2021-08-08:
Same here. Raspi Zero W.

lbt, 2021-08-16:
I had the same issue.

The SPI pin 10 works for a short while (about 1 second) and allows a few leds to change but then they go all white.
PWM on 18 with sudo works fine.

I found that doing some 'work' on the pi would make the LEDs flicker... turns out the cpu scaling was the issue as per the Pi 3 comments.

Quick test with

    echo 1000000 > /sys/devices/system/cpu/cpu0/cpufreq/scaling_min_freq

and SPI is working perfectly :slightly_smiling_face: 

I couldn't find an option for /boot/config.txt so to make it work on every boot I did:

    apt install sysfsutils
    echo /devices/system/cpu/cpu0/cpufreq/scaling_min_freq = 1000000 > /etc/sysfs.d/99-cpu-min.conf

francescofact, 2021-09-04:
Other solutions?
The echo mentioned didn't solve anything. I'm on Pi 4B 1GB

lbt, 2021-09-04:
The values for a Pi 4 are different - I don't know what's needed there (yet). 

    cat /sys/devices/system/cpu/cpu0/cpufreq/scaling_available_frequencies

to see the values available and then try using approptiate echos to set min and max in that directory to those values and see which ones work.

Old-Engineer, 2021-10-30:
Hi
I had similar problems using RPI on my pi4B. Studying the read me info gave the answer that worked for me.

Add the following lines to /boot/config.txt and reboot:
    core_freq=500
    core_freq_min=500
Good luck

lbt, 2022-01-07:
Changing the values in boot/config.txt has a very similar effect but needs a reboot to change. The kernel frequency governor is tweakable whilst running.

stephbellemare, 2024-09-22:
This item can be closed.  i resolved the issue as well by locking core_freq in config.txt