jgarff/rpi_ws281x #451: Dockerfile - Building the Shared Library in a Dockerfile

canadrian72, 2021-03-14:
I am facing an issue when trying to implement this library in a .net core application using Docker. If anyone has successfully built the shared C library (**ws2811.so**) in a Dockerfile, I'd love to see how it was done.

I am running a .net core application using a Dockerfile, and am trying to get this library to work. After following the [build instructions](https://github.com/kenssamson/rpi-ws281x-csharp/tree/master/lib#building-the-native-library), I am getting the following System.DllNotFoundException:

>  System.DllNotFoundException: Unable to load shared library 'ws2811.so' or one of its dependencies. In order to help diagnose loading problems, consider setting the LD_DEBUG environment variable: libws2811.so: cannot open shared object file: No such file or directory

So it seems as if the ws2811.so shared library is not found from the server, even though I've included in the Raspberry Pi's `/usr/lib` and `/usr/local/lib` directories. I have also confirmed that the ws2811.so library is installed correctly on the device, as the test from the repo ran successfully. Not sure where to go from here, hopefully there is someone who could help me on this. Thanks!

NOTE: Also reported in [Dockerfile - Unable to load shared library 'ws2811.so' #5](https://github.com/rpi-ws281x/rpi-ws281x-csharp/issues/5)

canadrian72, 2021-03-18:
## Solution
To any future readers, I've figured out how to include the shared library within a docker image. The following steps describe what needs to be done:
1. First build the library to obtain `ws2811.s0`, following the build instructions (as linked in original issue)
2. Include this library in the working directory of your docker file
3. Copy the shared library into the docker image using `COPY ws2811.so ./`

And it works like a charm! Check out my [Stackoverflow post](https://stackoverflow.com/questions/66627270/including-shared-c-library-so-in-a-net-core-docker-image) for more details.

Cheers

closed 2021-03-18 by canadrian72

qaniqani, 2022-06-20:
Hi @canadrian72 
I am having similar problem at operating system level. I'm not using Docker.
But I couldn't find a solution. I put the ws2811 file next to my working files and under usr/lib, usr/local/lib but it didn't work.
Are you still interested in the subject?
Regards.

canadrian72, 2022-06-20:
I'm not sure I could help you at an operating system level, but I would say the shared library is probably not in the right place or not configured correctly. 

I ended up using python as the installation is much easier. The library can easily be installed through pip and it is much more elegant. In docker I simply pip install the library and all works well. Maybe open a new issue or post to stack overflow with more specifics if you're looking for help.

qaniqani, 2022-06-21:
Thanks, I solved the problem. In fact, it can be said that it is the same problem you have in Docker.
After performing pip installation and configuration at OS level, I ran it successfully.
Thanks for replying, good luck.