jgarff/rpi_ws281x #454: TypeError when trying to run rpi_ws281x example

tmckay1, 2021-04-06:
Hi folks,

I have been using this library for a long time (since 2016/2017) and I haven't ran into an issue until now. I am trying to simply run the `strandtest.py` example in the `python/examples` directory. However I am getting the following error:
```python
pi@raspberrypi:~/scripts/rpi_ws281x/python/examples $ sudo python3 ./strandtest.py
Traceback (most recent call last):
  File "./strandtest.py", line 89, in <module>
    strip = Adafruit_NeoPixel(LED_COUNT, LED_PIN, LED_FREQ_HZ, LED_DMA, LED_INVERT, LED_BRIGHTNESS, LED_CHANNEL)
  File "/usr/local/lib/python3.7/dist-packages/rpi_ws281x/rpi_ws281x.py", line 94, in __init__
    ws.ws2811_channel_t_gamma_set(self._channel, gamma)
TypeError: in method 'ws2811_channel_t_gamma_set', argument 2 of type 'uint8_t *'
swig/python detected a memory leak of type 'ws2811_t *', no destructor found.
```
I have installed the driver with the following commands:
```
$ git clone https://github.com/jgarff/rpi_ws281x.git
$ cd rpi_ws281x
$ sudo apt-get install python-dev swig scons
$ sudo scons
$ cd python
$ sudo python3 setup.py build install
```
I am running the latest version of raspberry pi OS, I am also able to reproduce on Buster 2/14/2020. I have python 3.7:
```
pi@raspberrypi:~ $ uname -a
Linux raspberrypi 5.10.17-v7l+ #1403 SMP Mon Feb 22 11:33:35 GMT 2021 armv7l GNU/Linux
pi@raspberrypi:~ $ python3 --version
Python 3.7.3
```
I had to do something I usually don't have to do in order to import the `rpi_ws281x` module. I noticed that when I installed the driver with the above commands, my `dist-packages` folder is missing a file it usually has `rpi_ws281x.py` file:
```
pi@raspberrypi:~/scripts/rpi_ws281x/python/examples $ ll /usr/local/lib/python3.7/dist-packages/
total 12
-rw-r--r-- 1 root staff   42 Apr  2 00:48 easy-install.pth
drwxr-sr-x 4 root staff 4096 Apr  2 00:48 rpi_ws281x-1.0.0-py3.7-linux-armv7l.egg
```
In `easy-install.pth` we import `rpi_ws281x-1.0.0-py3.7-linux-armv7l.egg` which contains the modules for the driver. Typically, and this has been the case since I've used this library, after I install with the mentioned commands before there is an additional folder `rpi_ws281x` with the following contents:
```python
pi@raspberrypi:~ $ ll rpi_ws281x
total 20
-rw-r--r-- 1 pi pi  170 Mar 25 21:26 __init__.py
-rw-r--r-- 1 pi pi 7134 Mar 25 21:26 rpi_ws281x.py
pi@raspberrypi:~ $ cat rpi_ws281x/__init__.py
# New canonical package, to support `import rpi_ws281x`
from .rpi_ws281x import PixelStrip, Adafruit_NeoPixel, Color, ws
from _rpi_ws281x import *

__version__ = '4.2.4'
pi@raspberrypi:~ $ cat rpi_ws281x/rpi_ws281x.py
# Adafruit NeoPixel library port to the rpi_ws281x library.
# Author: Tony DiCola (tony@tonydicola.com), Jeremy Garff (jer@jers.net)
import _rpi_ws281x as ws
import atexit

... (more code)
```
Without this folder I would get a module not found error when trying `import rpi_ws281x`. So I manually added this folder and the file contents to the `dist-packages` folder, which allows me to import `rpi_ws281x` as expected. However, once I can import it I get the error that I logged before:
```python
pi@raspberrypi:~/scripts/rpi_ws281x/python/examples $ sudo python3 ./strandtest.py
Traceback (most recent call last):
  File "./strandtest.py", line 89, in <module>
    strip = Adafruit_NeoPixel(LED_COUNT, LED_PIN, LED_FREQ_HZ, LED_DMA, LED_INVERT, LED_BRIGHTNESS, LED_CHANNEL)
  File "/usr/local/lib/python3.7/dist-packages/rpi_ws281x/rpi_ws281x.py", line 94, in __init__
    ws.ws2811_channel_t_gamma_set(self._channel, gamma)
TypeError: in method 'ws2811_channel_t_gamma_set', argument 2 of type 'uint8_t *'
swig/python detected a memory leak of type 'ws2811_t *', no destructor found.
```
Any suggestions? Writing this out it just made me aware it could be the `scons` binary is on a newer version that contains this shortcoming. But before I dive into that, wondering if anyone here could give me some advice if it is apparent what is happening here

tmckay1, 2021-05-27:
Uninstall/reinstall worked with `sudo pip3 install rpi-ws281x`, thanks

closed 2021-12-03 by Gadgetoid

SaigeP, 2024-03-11:
I get this error message when installed the package on R Pi4: 
sudo apt-get install python3-rpi_ws281x
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package python3-rpi_ws281x

Any ideas how to fix this error? :)

Gadgetoid, 2024-03-11:
> sudo apt-get install python3-rpi_ws281x 

There is no apt package-

`pip install rpi_ws281x`

SaigeP, 2024-03-11:
Yeah, I have tried that as well. Not sure why it won't let me use the library... Iv'e tried on both Pi4 and Pi5. Any ideas why?

SaigeP, 2024-03-11:
pip install rpi_ws281x

This is what returns when I type in the above:
error: externally-managed-environment

× This environment is externally managed
╰─> To install Python packages system-wide, try apt install
    python3-xyz, where xyz is the package you are trying to
    install.
    
    If you wish to install a non-Debian-packaged Python package,
    create a virtual environment using python3 -m venv path/to/venv.
    Then use path/to/venv/bin/python and path/to/venv/bin/pip. Make
    sure you have python3-full installed.
    
    For more information visit http://rptl.io/venv

note: If you believe this is a mistake, please contact your Python installation or OS distribution provider. You can override this, at the risk of breaking your Python installation or OS, by passing --break-system-packages.
hint: See PEP 668 for the detailed specification.

For reference:
pip --version
pip 23.0.1 from /usr/lib/python3/dist-packages/pip (python 3.11)

SaigeP, 2024-03-20:
In case anyone is wondering, I was able to get some base level functionality (its not great, but kinda works) doing the following steps:
1. Download Debian Bullseye 32 bit
2. Install Python 3 in Terminal
3. Install Pip in Terminal
4. Install rpi_ws281x, then install RPi.GPIO, etc 
5. Then install VS Code