jgarff/rpi_ws281x #455: Can't seem to get the Strip.show() to work?

kittentakara, 2021-04-08:
Trying get the colour wipe to do it over and over, but it does not work, it just wipes once and then sits there until I end the program...Not sure if its bug or me being a newbie

def colorWipe(strip, Color, wait_ms=50, iterations=10, pixle_start=0, end_pixle=34):
    """Wipe color across display a pixel at a time."""
    for j in range(iterations):
        for i in range(pixle_start, end_pixle, 1):
            strip.setPixelColor(i, Color)
            strip.show()
            time.sleep(wait_ms/1000.0)

kittentakara, 2021-04-08:
Also does this have reference of all the commands and what they do?

kittentakara, 2021-04-09:
Here is the whole code 

#!/usr/bin/env python3
# rpi_ws281x library strandtest example
# Author: Tony DiCola (tony@tonydicola.com)
#
# Direct port of the Arduino NeoPixel library strandtest example.  Showcases
# various animations on a strip of NeoPixels.
 
import concurrent.futures
import time
from rpi_ws281x import *
import argparse
 
# LED strip configuration:
LED_COUNT      = 127      # Number of LED pixels.
LED_PIN        = 18      # GPIO pin connected to the pixels (18 uses PWM!).
#LED_PIN        = 10      # GPIO pin connected to the pixels (10 uses SPI /dev/spidev0.0).
LED_FREQ_HZ    = 800000  # LED signal frequency in hertz (usually 800khz)
LED_DMA        = 10      # DMA channel to use for generating signal (try 10)
LED_BRIGHTNESS = 255     # Set to 0 for darkest and 255 for brightest
LED_INVERT     = False   # True to invert the signal (when using NPN transistor level shift)
LED_CHANNEL    = 0       # set to '1' for GPIOs 13, 19, 41, 45 or 53
LEFTPIXLE      = [[0,36]] #Each part is start and end of each segement on the strip
MIDDLELEFT     = [[37,84]]
MIDDLERIGHT    = [[42,46],[53,57],[64,68],[75,79],[86,89]]
MIDDLEPIXLE    = [[37,89]]
RIGHTPIXLE     = [[90,126]]

threads = [] 

def colorWipe(strip, Color, wait_ms=50, iterations=10, pixle_start=0, end_pixle=34):
    """Wipe color across display a pixel at a time."""
    for j in range(iterations):
        for i in range(pixle_start, end_pixle, 1):
            strip.setPixelColor(i, Color)
            strip.show()
            time.sleep(wait_ms/1000.0)
          
            

RED = (255, 0, 0)
YELLOW = (255, 150, 0)
ORANGE = (255, 40, 0)
GREEN = (0, 255, 0)
TEAL = (0, 255, 120)
CYAN = (0, 255, 255)
BLUE = (0, 0, 255)
PURPLE = (180, 0, 255)
MAGENTA = (255, 0, 20)
WHITE = (255, 255, 255)

 
# Main program logic follows:
if __name__ == '__main__':
    # Process arguments
    parser = argparse.ArgumentParser()
    parser.add_argument('-c', '--clear', action='store_true', help='clear the display on exit')
    args = parser.parse_args()
 
    # Create NeoPixel object with appropriate configuration.
    strip = Adafruit_NeoPixel(LED_COUNT, LED_PIN, LED_FREQ_HZ, LED_DMA, LED_INVERT, LED_BRIGHTNESS, LED_CHANNEL)
    # Intialize the library (must be called once before other functions).
    strip.begin()
 
    print ('Press Ctrl-C to quit.')
    if not args.clear:
        print('Use "-c" argument to clear LEDs on exit')
 
    try:
 
        while True:
            with concurrent.futures.ThreadPoolExecutor() as executor: #Need this to run the executor.submit in each of these this allows more than one at time
                
                for position in MIDDLELEFT:   #Middle Pixel Group 
                    f1 = executor.submit(colorWipe, strip, Color(0 , 0, 255),pixle_start=position[0],end_pixle=position[1]) 
                for position in LEFTPIXLE: #Left Pixel group 
                    f2 = executor.submit(colorWipe, strip, Color(255, 0, 0),pixle_start=position[0],end_pixle=position[1])
                print ('theaterCase Press Ctl+C to exit.')
           
    except KeyboardInterrupt:
        if args.clear:
            colorWipe(strip, Color(0,0,0), 10)

Gadgetoid, 2021-12-03:
I'd suggest that rpi_ws281x and the Python bindings are *extremely not thread-safe* and venture `concurrent.futures.ThreadPoolExecutor()` is the source of your woes.

You should pare this back to a minimum viable example and - if you still have problems - raise an issue against the Python bindings here: https://github.com/rpi-ws281x/rpi-ws281x-python/issues

closed 2021-12-03 by Gadgetoid