jgarff/rpi_ws281x #459: I2s audio output with PWM led strip?

hyansuper, 2021-05-17:
I don't fully understand the [readme](https://github.com/jgarff/rpi_ws281x/#gpio-usage) about PWM and PCM. Will they interfere each other if I want to use them both.
I have a pi zero w. Can I use the [I2S MAX98357 amplifier](https://www.adafruit.com/product/3006) (pin18) to output audio and PWM (pin13) to control led strip? would they work at the same time?

thank you~

lmeysel, 2021-05-25:
I got nearly the same setup, Pi Zero W, i2s amplifier and ws2811 LED strip and got things running (I prefer using SPI, as it allows controlling the LEDs as non-root user, but also tested it with PWM).

Altough the song is playing "recognizable" while the LEDs are running, the problem is a crackling in sound appears every time I call render(). Unfortunately I did not find out why...

hyansuper, 2021-05-25:
Does the crack sound appear with SPI or PWM？
I didn't test SPI coz it's already in use，but PWM seems no problem

lmeysel, 2021-05-25:
It appeared in both, PWM and SPI. I thought it could have to do with mem speed or CPU, but it actually was because of the power-supply 🤦‍♂️

I used the same supply for LED and Amplifier (in parallel). While the LEDs changing the color, the load seems to be a bit higher, which makes the supply kind of "slightly instable", which resulted in a fluctuating bias point of the sound amplifier which again results in the crackling noise.

closed 2021-05-25 by hyansuper