jgarff/rpi_ws281x #460: Can't use PWM on Ubuntu 21.04

Mercotui, 2021-05-21:
I tried using the test application with the default GPIO 18 (PWM0) on my Raspbery Pi 3b, running Ubuntu 21.04.
However my LED strip stays off, and this is the console output:
```bash
$ sudo ./test -c
Can't open device file
: No such device or address
ws2811_init failed: Failed to create mailbox device
```
Looking into the code, it seems in `mailbox.c` the function `mbox_open` tries to open a device called `/dev/vcio`.
Sadly this doesn't exist on Ubuntu 21.04.
```bash
$ ls /dev/vcio
ls: cannot access '/dev/vcio': No such file or directory
```
The backup mechanism to create a node in `/tmp/` also clearly fails, but this is beyond my understanding :shrug: 

### My fix:
In [another](https://github.com/jgarff/rpi_ws281x/issues/293#issuecomment-384188030) issue I saw the recommendation to use SPI, on GPIO 10 instead.
Trying that:
```
$ ./test --gpio 10 -c
```
The screen lights up in a rainbow, works wonderfully!

Mercotui, 2021-05-21:
leaving mostly for prosperity, thank you to the contributors for making this fine library :ok_hand:

progman32, 2022-01-05:
Just another headsup. I also ran into the mailbox issue on my B Rev. 2 in Debian 10, but was additionally unable to get SPI working. It appears the SPI driver doesn't load properly, despite configuring it in config.txt. https://github.com/raspberrypi/linux/issues/526

jwillikers, 2022-04-18:
`/dev/vcio` does not exist on Fedora either on the Raspberry Pi CM4.

isidroas, 2024-07-12:
Here I am using Raspberry Pi 4 with an image from https://raspi.debian.net/tested-images/

Devices `/dev/vcio` and `/dev/spidev0.0` do not exist.

However I tried [WiringPi/examples/pwm.c](https://github.com/WiringPi/WiringPi/blob/813c3130be372841d84653dc8a1c2bf6e500d0e2/examples/pwm.c) and it works.

Is there another way to control PWM?

airtower-luna, 2025-01-29:
I'm getting the same error message (Python exception added because I'm using [rpi-ws281x-python](https://github.com/rpi-ws281x/rpi-ws281x-python)) after switching from RPi kernel to mainline:

```
Can't open device file
: No such device or address
Traceback (most recent call last):
  File "/root/test-neopixel.py", line 135, in <module>
    strip.begin()
  File "/usr/lib/python3.12/site-packages/rpi_ws281x/rpi_ws281x.py", line 143, in begin
RuntimeError: ws2811_init failed with code -9 (Failed to create mailbox device)
Segmentation fault
```

Is there any supported way of running rpi-ws281x with a mainline kernel, or a recommended alternative? I noticed the `/dev/vcio` device with RPi kernel and the `/dev/vchiq` present on mainline have the same major and minor numbers, but I'm not sure if that says anything.