jgarff/rpi_ws281x #462: 1440 leds 25fps

artcontr, 2021-05-24:
Hi. I'm trying to run a matrix of 10x144 rgbw SK6812 leds. I need to play animations on them (maybe gifs or small video loops). Do you think your library will allow me to do so ? Thanks a lot :-)

InstanceGaming, 2021-07-25:
You can easily write a test program where you write random data to your matrix and then time the execution of `show()`. This will definitively prove weather your environment can handle the refresh rate you desire. If the benchmark yields less than 40ms, you should be able to make it work at 25fps. However, additional time should be allocated for whatever processing you need to do first in order to generate a frame, like video decoding (which is more likely to cause bottlenecks).

I myself have 1944 pixels to drive and my rudimentary benchmarking (even through the Python binding) yielded about ~4ms execution time calling `show()` on a Raspberry Pi 4B+.

For timing that function call in C (on Linux), [start here](https://stackoverflow.com/questions/6749621/).

Gadgetoid, 2022-01-25:
https://github.com/jgarff/rpi_ws281x/pull/468 may be relevant to your interests.