jgarff/rpi_ws281x #465: Cannot run on Pi 4 without sudo - Can't open /dev/mem: Permission denied -  Failed to setup lights driver, result code: WS2811_ERROR_MMAP

Sciss, 2021-06-18:
I am using the Java / JNI wrapper. I already tried some suggestions like adding user pi to group gpio, and chown and chmod on /dev/mem, this doesn't change the problem:

```
16:43:59.142 [main] INFO  com.github.mbelling.ws281x.Ws281xLedStrip - Native library loaded
Can't open /dev/mem: Permission denied
!! Failed to initialize WS2811:
java.lang.RuntimeException: Failed to setup lights driver, result code: WS2811_ERROR_MMAP
	at com.github.mbelling.ws281x.Ws281xLedStrip.init(Ws281xLedStrip.java:234)
	at com.github.mbelling.ws281x.Ws281xLedStrip.<init>(Ws281xLedStrip.java:98)
```

As my application needs to read and write user files, running it as root is not an option. Please help find a fix to this problem. Thanks

cp2004, 2021-06-27:
You can use SPI instead of PWM. You can't use PWM without root.

Sciss, 2021-06-27:
I see. My workaround is now to run the PWM control in a separate process that is controlled via OSC input, so only that process has to run under `sudo`.

closed 2021-06-27 by Sciss