jgarff/rpi_ws281x #466: 12 Led NeoPixel Ring Raspberry Pi 4

bobster316, 2021-07-02:
I am trying to run an example script in rpi_ws281x 
When running the following command -
sudo python3 /home/pi/rpi_ws281x/python/examples/strandtest.py

I get the following error -
Traceback (most recent call last) : 
  File "strandtest.py", line 9. in <module>
       from rpi_ws281x import "
ModuleNotFoundError: No module named 'rpi_ws281x'

Can anyone please help to resolve

ranjithp97, 2021-07-10:
Hi @bobster316, I am assuming that you have installed pi OS recently where python2 version of library rpi_ws281x is a part of installation files. But if you want to run the python file in python3 version you need to update that firmware by using the below command on terminal window

sudo pip3 install rpi_ws281x

It worked for me, hope it works for you

Gadgetoid, 2021-10-05:
Note that the Python code in this repository is deprecated. The python bindings are here: https://github.com/rpi-ws281x/rpi-ws281x-python

closed 2021-10-05 by Gadgetoid