jgarff/rpi_ws281x #472: Cannot Build setup.py on RP4

robofred, 2021-10-19:
Newbie here - was able to clone/build/install and run strandtest.py on RP4-first but when I tried running the exact same steps on my RP4-second, I get the following error output:

`Extracting in /tmp/tmpygytrjlh
Now working in /tmp/tmpygytrjlh/setuptools-5.7
Building a Setuptools egg in /home/pi/neopix/rpi_ws281x/python
Traceback (most recent call last):
  File "setup.py", line 19, in <module>
    exec(init_file.read(), command_ns)
  File "<string>", line 11, in <module>
  File "/tmp/tmpygytrjlh/setuptools-5.7/setuptools/__init__.py", line 12, in <module>
    from setuptools.extension import Extension
  File "/tmp/tmpygytrjlh/setuptools-5.7/setuptools/extension.py", line 7, in <module>
    from setuptools.dist import _get_unpatched
  File "/tmp/tmpygytrjlh/setuptools-5.7/setuptools/dist.py", line 18, in <module>
    import pkg_resources
  File "/tmp/tmpygytrjlh/setuptools-5.7/pkg_resources.py", line 1520, in <module>
    register_loader_type(importlib_bootstrap.SourceFileLoader, DefaultProvider)
AttributeError: module 'importlib._bootstrap' has no attribute 'SourceFileLoader'
/home/pi/neopix/rpi_ws281x/python/setuptools-5.7-py3.7.egg
Traceback (most recent call last):
  File "/home/pi/neopix/rpi_ws281x/python/ez_setup.py", line 142, in use_setuptools
    pkg_resources.require("setuptools>=" + version)
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 900, in require
    needed = self.resolve(parse_requirements(requirements))
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 786, in resolve
    raise DistributionNotFound(req, requirers)
pkg_resources.DistributionNotFound: The 'setuptools>=5.7' distribution was not found and is required by the application

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "setup.py", line 4, in <module>
    use_setuptools()
  File "/home/pi/neopix/rpi_ws281x/python/ez_setup.py", line 145, in use_setuptools
    return _do_download(version, download_base, to_dir, download_delay)
  File "/home/pi/neopix/rpi_ws281x/python/ez_setup.py", line 120, in _do_download
    _build_egg(egg, archive, to_dir)
  File "/home/pi/neopix/rpi_ws281x/python/ez_setup.py", line 69, in _build_egg
    raise IOError('Could not build the egg.')
OSError: Could not build the egg.
`
I'm very new to the Linux operating system. I do know there's very little difference between my dev platform (RP4-first) and my production (RP4-second) platform. Any ideas whats happening here?

robofred, 2021-10-19:
I tried to install rpi_ws281x but it turned out that my production machine did not have pip3 installed.

Once I installed pip3,

`sudo apt-get install python3-pip`

I was able to build setup.py correctly... and from there everything worked great. 

Thanks a lot

/F

Gadgetoid, 2021-11-02:
The Blinka/CircuitPython lib is unrelated to rpi_ws281x- albeit we did once have a "neopixel" library as part of the Python distribution that we dropped at their request- they wanted to build cross-platform "neopixel" support, using rpi_ws281x but without the two packages conflicting.

There should be no need for it!

closed 2021-12-03 by Gadgetoid