jgarff/rpi_ws281x #475: Doesn't support Pi Zero 2

jwalker55, 2021-11-02:
Are there plans to add support for the Pi Zero 2?

When trying to setup a neopixel node in node-red, it doesn't work. When testing via console, I get:

```
 ./neopix 3 40 pixels 100 true 0 35
Traceback (most recent call last):
  File "./neopix.py", line 171, in <module>
    strip.begin()
  File "/usr/local/lib/python3.7/dist-packages/rpi_ws281x/rpi_ws281x.py", line 131, in begin
    raise RuntimeError('ws2811_init failed with code {0} ({1})'.format(resp, str_resp))
RuntimeError: ws2811_init failed with code -3 (Hardware revision is not supported)
./neopix: line 7:  2437 Segmentation fault      sudo $python_cmd -u $BASEDIR/neopix.py $@
```
The relevant line: `RuntimeError: ws2811_init failed with code -3 (Hardware revision is not supported)`

jwalker55, 2021-11-02:
Disregard. I see the update from 3 days ago. My version must be out of date. Apologies.

closed 2021-11-02 by jwalker55

Gadgetoid, 2021-11-02:
I have not released a new Python package yet, actually :grimacing:

jwalker55, 2021-11-02:
I will wait patiently. :)

Gadgetoid, 2021-11-02:
Should be supported in - https://pypi.org/project/rpi-ws281x/4.3.1/

jwalker55, 2021-11-02:
Works great. Thanks!

Richard238, 2021-11-03:
Trying Unicornhat on Pi Zero 2 and got the code -3 error  ran `pip install rpi-ws281x==4.3.1`
Now get code -5 error.