jgarff/rpi_ws281x #478: undefined reference to 'ws2811_...'

R3tr0BoiDX, 2021-12-02:
I did installed the libbrary as descriped in the README by doing

`mkdir build
cd build
cmake -D BUILD_SHARED=ON -D BUILD_TEST=OFF ..
cmake --build .
sudo make install`

It says
`[100%] Built target ws2811
Install the project...
-- Install configuration: ""
-- Installing: /usr/local/lib/libws2811.so
-- Installing: /usr/local/include/ws2811/ws2811.h
-- Installing: /usr/local/include/ws2811/rpihw.h
-- Installing: /usr/local/include/ws2811/pwm.h
-- Installing: /usr/local/include/ws2811/clk.h
-- Installing: /usr/local/include/ws2811/dma.h
-- Installing: /usr/local/include/ws2811/gpio.h
-- Installing: /usr/local/include/ws2811/mailbox.h
-- Installing: /usr/local/include/ws2811/pcm.h
-- Installing: /usr/local/lib/pkgconfig/libws2811.pc
`

However, if I want to compile my code with `gcc main.c -o test`, I get `main.c:(.text+0x18): undefined reference to `ws2811_fini'` for example.

My `main.c`:
`#include <ws2811/ws2811.h>

int main(int argc, char const *argv[])
{
    return 0;
}`

Gadgetoid, 2021-12-03:
The header file "declares" all the functions you're using, but the object file or library is responsible for "defining" them.

You need to add `-lws2811` to instruct gcc to link against the ws2812 library and include the definitions of `ws2811_fini`. In the simplest sense this either smooshes your code and the library code together into a single executable (static linking) or allows gcc to create a binary that knows where to find the library file dynamic linking).

So:

```
gcc main.c -lws2811 -o test
```

KC5-BP, 2021-12-08:
With your solution, I still have an undefined reference to pow, with this output :
$ gcc main.c -lws2811 -o test
/usr/bin/ld: /usr/local/lib/libws2811.a(ws2811.c.o): in function `ws2811_set_custom_gamma_factor':
ws2811.c:(.text+0x2934): undefined reference to `pow'
collect2: error: ld returned 1 exit status

And looking in the ws2811.c file, the #include <math.h> is called.

And I upgraded my system to be sure. So is it a problem of my gcc of my line is still missing something ?

(To be sure, I've built the binaries and having the header, they're in /usr/local/include/ws2811

And the ls of my folder is as follow ; 
$ ls
main.c  version.h

KC5-BP, 2021-12-08:
In advance, if you encounter that issue :
$ gcc main.c -lws2811 -o test
/usr/bin/ld: /usr/local/lib/libws2811.a(ws2811.c.o): in function `ws2811_set_custom_gamma_factor':
ws2811.c:(.text+0x2934): undefined reference to `pow'
collect2: error: ld returned 1 exit status

Just link the math lib with the following : 
$ gcc main.c -lws2811 -lm -o test