jgarff/rpi_ws281x #483: Raspberry Pi 4 Model B Rev 1.5 not supported?

Ceberus132, 2022-01-19:
Hello,

I assume, that the Raspberry Pi 4 Model B Rev 1.5 is not yet supported, since the rpihw.c also just lists to the 1.4 Rev (I also get the "ws2811_init failed with code -3 (Hardware revision is not supported)" RuntimeError), am I right?

If so, will the Rev 1.5 be supported in the future, if yes, when?^^

Have a nice Day or Night =D

Tirenoth, 2022-01-20:
Thank you! This worked like a charm.

One modification  - since Easy Install is deprecated, running `sudo python3 setup.py install` didn't work correctly for me.

Instead, I needed to run `sudo pip3 install /path/to/home/rpi-ws281x-python/library`

Gadgetoid, 2022-01-24:
If you find a new, unsupported revision you should also raise a PR to support it :smile:

Tirenoth, 2022-01-24:
That didn't even occur to me while I was looking for a solution. Now it seems quite silly not to! :D

PR opened - https://github.com/jgarff/rpi_ws281x/pull/484

Macleykun, 2022-02-24:
> If you find a new, unsupported revision you should also raise a PR to support it 😄

Heya, i believe i have a new unsupported revision:
Hardware        : BCM2835
Revision        : c03115
Serial          : 100000002fcc88bc
Model           : Raspberry Pi 4 Model B Rev 1.5

I'm sadly completely new to this (using a rgb ring, wiring and low level code) but what can i give of information to support this board aswell for my wrgb ring 16 led?

bbeesley, 2022-02-24:
> Heya, i believe i have a new unsupported revision: Hardware : BCM2835 Revision : c03115 Serial : 100000002fcc88bc Model : Raspberry Pi 4 Model B Rev 1.5
>
> I'm sadly completely new to this (using a [...]

I've opened a PR for you using the details you provided #492 
Am I right in thinking this is the 4GB pi 4?

Macleykun, 2022-02-25:
> I've opened a PR for you using the details you provided #492 Am I right in thinking this is the 4GB pi 4?

Yes sir!

michaelacosta-rpi, 2022-03-02:
I think I have a new unsupported version as well. 
Hardware	: BCM2835
Revision	: a03115
Serial		: 10000000c35ef311
Model		: Raspberry Pi 4 Model B Rev 1.5

I tried doing the steps above with no success. Any suggestions or help?

Macleykun, 2022-03-02:
Pho’s much memory?

Met vriendelijke groet,

Wesley de Vree

michaelacosta-rpi, 2022-03-02:
1GB RAM, sorry I left that out

Macleykun, 2022-03-04:
> Hello,
> 
> if u try to run a Script from the Repo and you get the RuntimeError: "ws2811_init failed with code -3 (Hardware revision is not supported)" I found a solution for u =D
> 
> First open your shell, type in: `cat /proc/cpuinfo` or go to the Explorer, into "home/proc/cpuinfo" and open the File. At the Bottom will be a Part which looks like:
> 
> ```
> Hardware	: BCM2711
> Revision	: b03115
> Serial		: 100000000ed9a28c
> Model		: Raspberry Pi 4 Model B Rev 1.5
> ```
> 
> Now open the shell, type in `nano (or vim) rpi-ws281x/library/lib/rpihw.c` and search for your Model. If you cant find it, we will follow this Guide: [rpi-ws281x/rpi-ws281x-python#56 (comment)](https://github.com/rpi-ws281x/rpi-ws281x-python/issues/56#issuecomment-753320723)
> 
> Open the shell, first you uninstall the repo with `sudo pip3 uninstall rpi_ws281x`,
> 
> next you clone the git with `git clone --recurse-submodules https://github.com/rpi-ws281x/rpi-ws281x-python`,
> 
> after that you run `nano (or vim) rpi-ws281x-python/library/lib/rpihw.c`,
> 
> then you add:
> 
> ```
> { 
>      .hwver = 0xb03115, (Based on the Revision found in cpuinfo)
>      .type = RPI_HWVER_TYPE_PI4, 
>      .periph_base = PERIPH_BASE_RPI4, 
>      .videocore_base = VIDEOCORE_BASE_RPI2, 
>      .desc = "Pi 4 Model B - 2GB v1.5" (Based on the Model in cpuinfo)
> },
> ```
> 
> after you made the changes, you change the Directory with `cd rpi-ws281x-python/library/`,
> 
> the last Command will be: `sudo python3 setup.py install`
> 
> I hope that helped, have a nice Day or Night =D

@michaelacosta-rpi  follow the above quote and ya should be able to open a pr here (so everyone can use it it’s that revision!)

Macleykun, 2022-03-11:
> Unfortunately I followed those steps and get the same error

sorry for the late reply
so, just to double check, adding this:

```
    {
        .hwver = 0xa03115,
        .type = RPI_HWVER_TYPE_PI4,
        .periph_base = PERIPH_BASE_RPI4,
        .videocore_base = VIDEOCORE_BASE_RPI2,
        .desc = "Pi 4 Model B - 1GB v1.5"
    },
```
put it in this line: https://github.com/bbeesley/rpi_ws281x/blob/e1df895b7d816a7b0352ca267dc210bf8dbe5752/rpihw.c#L138

so that you get:
(from line 138))
```
    },
    {
        .hwver = 0xa03115,
        .type = RPI_HWVER_TYPE_PI4,
        .periph_base = PERIPH_BASE_RPI4,
        .videocore_base = VIDEOCORE_BASE_RPI2,
        .desc = "Pi 4 Model B - 1GB v1.5"
    },
    {
        .hwver = 0xb03115,
        .type = RPI_HWVER_TYPE_PI4,
        .periph_base = PERIPH_BASE_RPI4,
        .videocore_base = VIDEOCORE_BASE_RPI2,
        .desc = "Pi 4 Model B - 2GB v1.5"
    },
```
and doing a recompile and then running with sudo the test script again still gives the error?
it shouldn't, i checked the others and what was changed to support my board, and it all should check out

marcuszerep, 2022-04-12:
If it is helpful, another config data for the 8gb...

Hardware	: BCM2711
Revision	: d03115
Serial		: 10000000bc84d13b
Model		: Raspberry Pi 4 Model B Rev 1.5

marcuszerep, 2022-05-03:
All 4 versions of Raspberry Pi 4 Model B Rev 1.5 -- 1gb 2gb 4gb 8gb:

    {
        .hwver = 0xa03115,
        .type = RPI_HWVER_TYPE_PI4,
        .periph_base = PERIPH_BASE_RPI4,
        .videocore_base = VIDEOCORE_BASE_RPI2,
        .desc = "Pi 4 Model B - 1GB v1.5"
    },
    {
        .hwver = 0xb03115,
        .type = RPI_HWVER_TYPE_PI4,
        .periph_base = PERIPH_BASE_RPI4,
        .videocore_base = VIDEOCORE_BASE_RPI2,
        .desc = "Pi 4 Model B - 2GB v1.5"
    },
    {
        .hwver = 0xc03115,
        .type = RPI_HWVER_TYPE_PI4,
        .periph_base = PERIPH_BASE_RPI4,
        .videocore_base = VIDEOCORE_BASE_RPI2,
        .desc = "Pi 4 Model B - 4GB v1.5"
    },
    {
        .hwver = 0xd03115,
        .type = RPI_HWVER_TYPE_PI4,
        .periph_base = PERIPH_BASE_RPI4,
        .videocore_base = VIDEOCORE_BASE_RPI2,
        .desc = "Pi 4 Model B - 8GB v1.5"
    },

Notalso, 2022-05-28:
> doing a recompile and then running with sudo the test script again still gives the error? it shouldn't, i checked the others and what was changed to support my board, and it all should check out
@Macleykun or @michaelacosta-rpi , were you able to solve this issue for the RPi 4 Model B 1 GB?

I have the same issue and have tried to use this method to solve the problem, but have not succeeded. I would appreciate if anyone could point out what I've missed so that I can start playing with my set of NeoPixels.

I started with this guide to use NeoPixels with a Raspberry Pi (though I have repeatedly uninstalled and installed since then):
https://learn.adafruit.com/neopixels-on-raspberry-pi/python-usage
sudo pip3 install rpi_ws281x adafruit-circuitpython-neopixel
sudo python3 -m pip install --force-reinstall adafruit-blinka

I have performed OS updates listed as prerequisites here:
https://learn.adafruit.com/circuitpython-on-raspberrypi-linux/installing-circuitpython-on-raspberry-pi

Code I'm trying to run:
```
import board
import neopixel
pixels = neopixel.NeoPixel(board.D18,30)
pixels[0] = (255,0,0)
```

My stack trace:
```
Traceback (most recent call last):
  File "/home/pi/Documents/NeoPixels/Demo.py", line 4, in <module>
    pixels[0] = (255,0,0)
  File "/usr/local/lib/python3.7/dist-packages/adafruit_pixelbuf.py", line 312, in __setitem__
    self.show()
  File "/usr/local/lib/python3.7/dist-packages/adafruit_pixelbuf.py", line 204, in show
    return self._transmit(self._post_brightness_buffer)
  File "/usr/local/lib/python3.7/dist-packages/neopixel.py", line 180, in _transmit
    neopixel_write(self.pin, buffer)
  File "/home/pi/.local/lib/python3.7/site-packages/neopixel_write.py", line 38, in neopixel_write
    return _neopixel.neopixel_write(gpio, buf)
  File "/home/pi/.local/lib/python3.7/site-packages/adafruit_blinka/microcontroller/bcm283x/neopixel.py", line 83, in neopixel_write
    "ws2811_init failed with code {0} ({1})".format(resp, message)
RuntimeError: ws2811_init failed with code -3 (Hardware revision is not supported)
```

My output of cat /proc/cpuinfo: (Raspberry Pi 4 Model B 1 GB - a03115)
```
Hardware	: BCM2711
Revision	: a03115
Serial		: 1000000030ded1c1
Model		: Raspberry Pi 4 Model B Rev 1.5
```

I ran the github command to download code to this directory:
> /home/pi/Documents/NeoPixels/rpi_ws281x/rpi-ws281x-python/library/lib

Inside rpihw.c on lines 138-144 I already see the appropriate values for my hardware model:
```
    {
        .hwver = 0xa03115,
        .type = RPI_HWVER_TYPE_PI4,
        .periph_base = PERIPH_BASE_RPI4,
        .videocore_base = VIDEOCORE_BASE_RPI2,
        .desc = "Pi 4 Model B - 1GB v1.5"
    }
```
	
So, I navigated to the \library directory and ran the command 
> sudo python3 setup.py install
This appeared to compile & install, but the issue was not resolved

I also tried the top commenter's command saying easy install is deprecated with no effect:
> sudo pip3 install .
This also appeared to run properly, but did not solve the issue.

So, I wanted to make sure the version I had pulled was the one being installed, so I tried uninstalling with this command:
> sudo pip3 uninstall rpi_ws281x
> sudo pip3 uninstall adafruit-circuitpython-neopixel

I believe this worked, because my stack trace changes to being unable to import NeoPixel:
Traceback (most recent call last):
```
  File "/home/pi/Documents/NeoPixels/Demo.py", line 2, in <module>
    import neopixel
ModuleNotFoundError: No module named 'neopixel'
```

I then try the above python install again:
> sudo pip3 install .
> sudo pip3 install adafruit-circuitpython-neopixel

Is there any way to look through the /site-packages/ and /dist-packages/ to find the rpihw.c file and make sure I've have an appropriate setup? I tried poking through the .py files in the stack trace but I'm not particularly experienced in python, or how it interacts with c.

Also, since the code imported from github apparently required no modifications, I would expect the command 
> sudo pip3 install rpi_ws281x
to install a functioning version, but it does not. What else am I missing?

marcuszerep, 2022-06-27:
I'd like to contribute to the resolution of this issue, but I am not a C/compiled code developer. I've gathered the needed json code https://github.com/jgarff/rpi_ws281x/issues/483#issuecomment-1115780466 for https://github.com/bbeesley/rpi_ws281x/blob/e1df895b7d816a7b0352ca267dc210bf8dbe5752/rpihw.c#L138

I can also DEV/UAT test for an 8GB v1.5 Pi. 

Also willing to hire an existing contributor to get this resolved. Would hire a fiver person to make the code change and compile, but I would rather compensate an existing contributor. Thanks all

marcuszerep, 2022-07-11:
For dependent libraries, although the issues remains open, this appears to be updated in the code.
https://github.com/jgarff/rpi_ws281x/blob/master/rpihw.c#L139

Raspberry Pi 4 Model B Rev 1.5:
1gb - https://github.com/jgarff/rpi_ws281x/commit/9d74e4786a68b5ce3bc73c24286d0e533d8ea315
2gb - https://github.com/jgarff/rpi_ws281x/commit/d8f211eb8e6cfce23e60c2a99f47115642c5d7be
4gb - https://github.com/jgarff/rpi_ws281x/commit/d5689872956852bdcd9518fc7394b90e8665f269
8gb - https://github.com/jgarff/rpi_ws281x/commit/c412b65425f4699b24175acf21fa45a244527910

Macleykun, 2022-08-17:
Any update on this one :) ?

marcuszerep, 2022-08-17:
This issue should be closed out. 

I can confirm I had someone compile the library and I've tested on Raspberry Pi 4 Model B Rev 1.5 8gb & 4gb. If you are still having an issue, you are either 1) using an older build, 2) using a library which is dependent on this with an older build (they need to update their dependency on this), or 3) did not compile the library properly or with the most recent build.

ayashjorden, 2022-09-02:
Hi all,
I'm trying to drive a ws2811 strip using various Node.JS libs, currently using `"@gbkwiatt/node-rpi-ws281x-native"`.
OS is Ubuntu 18.04.2

The board I have is [LibreComputer LePotator](https://www.libre.computer/products/s905x/), `cat /proc/cpuinfo`:
```
libre@libre-computer:~/src/myproject$ cat /proc/cpuinfo
processor       : 0
BogoMIPS        : 48.00
Features        : fp asimd evtstrm aes pmull sha1 sha2 crc32 cpuid
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 1
BogoMIPS        : 48.00
Features        : fp asimd evtstrm aes pmull sha1 sha2 crc32 cpuid
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 2
BogoMIPS        : 48.00
Features        : fp asimd evtstrm aes pmull sha1 sha2 crc32 cpuid
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 3
BogoMIPS        : 48.00
Features        : fp asimd evtstrm aes pmull sha1 sha2 crc32 cpuid
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4
```

Can anyone advise on what would be the parameters for `rpihw.c` ? or if there's another command to get the Hardware/Revision/Serial/Model data?

Thanks

Macleykun, 2025-09-07:
@Ceberus132 can you close this issue :) ?

closed 2025-09-07 by Ceberus132