jgarff/rpi_ws281x #487: Neopixel Ring 12 Led - Raspberry Pi OS 64 Bit

bobster316, 2022-02-07:
I am having issues trying to setup the Neopixel 12 Led Ring with raspberry pi os 64 bit.
The setup works fine with raspberry pi 32 bit os

I am using the following commands 

sudo pip3 uninstall rpi_ws281x 
sudo pip3 install rpi_ws281x
sudo pip3 install adafruit-blinka
sudo apt-get install python-dev git scons swig 
git clone https://github.com/jgarff/rpi_ws281x.git
cd rpi_ws281x
sudo scons

There should be a python folsub in rpi_ws281x which is missing
These are the remaining commands which subsequently fail 

cd python
sudo python3 setup.py build

Thanks for any help
Best Regard 
Rav

Gadgetoid, 2022-02-07:
> There should be a python folsub in rpi_ws281x which is missing
> These are the remaining commands which subsequently fail

This has been deprecated for over four years now and recently deleted.

It's replaced by the Python bindings and examples here: https://github.com/rpi-ws281x/rpi-ws281x-python

The Python library can be installed with: `sudo python3 -m pip install rpi_ws281x`

Macleykun, 2022-03-03:
Assume this can be closed as the user gaan responded in a month?