jgarff/rpi_ws281x #491: same issue on REV 1.5

PandaFondateur, 2022-02-24:
Hey, 

Just unpacked my branded new adeept robot and during webserver connexion it prompt "Hardware version not supported" . 

my Pi is the following : 

Hardware        : BCM2711
Revision        : c03115
Serial          : 10000000a9c50773
Model           : Raspberry Pi 4 Model B Rev 1.5

How to fix it please ! :)

Macleykun, 2022-02-24:
believe i have this issue aswell but with this https://www.reddit.com/r/adafruit/comments/t07qgq/comment/hy8clwq/?utm_source=share&utm_medium=web2x&context=3

PandaFondateur, 2022-02-28:
> it prompt : "RuntimeError: ws2811_init failed with code -3 (Hardware
> revision is not supported)" .
> you have the 4GB aswell btw?

yes :/

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Garanti
sans virus. www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

Macleykun, 2022-02-28:
> yes :/ <https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail> Garanti sans virus. [www.avast.com](http://www.avast.com) [...]

https://github.com/jgarff/rpi_ws281x/pull/492

If you want, make sure that you installed the basic pips from adafruit (forgot which once’s they are but if needed I can check those at home later today)

clone this repo and make the edit from that pull. If ya then follow the build steps in the readme you can temporarily get it to work and can try with the demo script if your leds work :)

really hope that pull gets merged soon!

Macleykun, 2022-03-02:
> […](#)

if you clone it now, no changes are needed only need to follow the build instructions and run the demo!

marcuszerep, 2022-05-03:
Duplicate of #483