jgarff/rpi_ws281x #503: Raspberry Pi Zero 2 W - Hardware revision is not supported

codac, 2022-08-31:
Running your library with a Raspberry Pi Zero 2 W results in:
RuntimeError: ws2811_init failed with code -3 (Hardware revision is not supported).

Hardware: BCM2835
Revision: 902120
Model: Raspberry Pi Zero 2 W Rev 1.0

Could you please add support for this Pi Revision?

akumpf, 2022-09-01:
I think this may be related to issue https://github.com/jgarff/rpi_ws281x/issues/502. 

Essentially, if the warranty bit gets flipped, then the library can't find the board. This _shouldn't_ happen, but the code that checks for the board version isn't masking off the warranty bit. It's just 2 lines of code, and it's already being applied for 32-bit boards, but for 64 bit boards it was somehow left out.

codac, 2022-09-01:
Yeah might be, could you please add that bit of code?

DaddyW123, 2023-02-20:
I just tried this library on a Mango Pi MQ-Quad, which is supposed to be an alternative to the Pi Zero 2 and runs at 64 bit.
I got the same error as above. Hopefully it's just a 64 bit error as mentioned, and as soon as the two lines of code are included it will work on this board as well?

closed 2023-05-16 by Gadgetoid