jgarff/rpi_ws281x #507: "ws2811_init failed: Hardware revision is not supported" on Banana Pi BPI-M2 ZERO

FerrariAndrea, 2022-10-30:
Hi! 
I destroy my Raspberry Pi Zero, so I buy a Banana Pi BPI-M2 ZERO and try to run my old Node.js code that was on the Raspberry.

I'm using the npm: `rpi-ws281x^1.0.36` and I receive the error:
`Error: configure(): ws2811_init() failed: Hardware revision is not supported`

So I download this repository and follow the instructions, after some build problems, I finally run the "test" file and I received the same error:
`
sudo ./test
ws2811_init failed: Hardware revision is not supported
`

Banana Pi information:
` cat /proc/cpuinfo`
```
processor       : 0
model name      : ARMv7 Processor rev 5 (v7l)
BogoMIPS        : 22.85
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xc07
CPU revision    : 5

processor       : 1
model name      : ARMv7 Processor rev 5 (v7l)
BogoMIPS        : 22.85
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xc07
CPU revision    : 5

processor       : 2
model name      : ARMv7 Processor rev 5 (v7l)
BogoMIPS        : 22.85
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xc07
CPU revision    : 5

processor       : 3
model name      : ARMv7 Processor rev 5 (v7l)
BogoMIPS        : 22.85
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xc07
CPU revision    : 5

Hardware        : sun8iw7
Revision        : 0000
Serial          : 649079310810582c0454
```
`cat /etc/os-release`
```
PRETTY_NAME="Raspbian GNU/Linux 9 (stretch)"
NAME="Raspbian GNU/Linux"
VERSION_ID="9"
VERSION="9 (stretch)"
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
```

How I can resolve that error? there is some step that I need to do in order to configure the gipos before running the scripts? 
There is a way to have more information about that error?

Thx :)

jgarff, 2022-10-30:
Unfortunately, as listed in the wiki, only the Raspberry Pi is supported.
The Banana Pi doesn't have the same chip hardware blocks to support this
feature.

FerrariAndrea, 2022-10-30:
Hi guys, I'm trying to add this Banana Pi to the configurations, but I'm missing something.
I add the model here: https://github.com/FerrariAndrea/rpi_ws281x/commit/117d3d73584d8340866c706c93714da8ef5e96fc
and now it is working, but I received a new error:
```
sudo ./test -s grb
Can't open device file
: No such device or address
ws2811_init failed: Failed to create mailbox device
```

I research about that and I discover that in this Banana PI there is only the PWM1 ad pin 7, but I can't use it:
```
sudo ./test -s grb -g 7
Gpio 7 is illegal for LED channel 0
ws2811_init failed: Selected GPIO not possible
```
This is because (I think) in the code there are many hard-coded checks that don't allow the use of it.
I'm trying to remove all these checks in order to test it, but still receive "illegal" error for now

closed 2023-03-06 by Gadgetoid