jgarff/rpi_ws281x #508: RuntimeError: ws2811_init failed with code -3 (Hardware revision is not supported) - Raspberry Pi 3 Model B Rev 1.0

Jamie-Holland, 2022-11-23:
Added this to the rpihw.c file: 
    {

        .hwver = 0xa02080
        .type = RPI_HWVER_TYPE_PI2,
        .periph_base = PERIPH_BASE_RPI2,
        .videocore_base = VIDEOCORE_BASE_RPI2,
        .desc = "Raspberry Pi 3 Model B Rev 1.0",

    },

which when running : 

cd rpi-ws281x-python/library/
sudo python3 setup.py install

results in:

lib/rpihw.c:449:2: error: request for member ‘type’ in something not a structure or union
  449 |  .type = RPI_HWVER_TYPE_PI2,
      |  ^

This is being used on a Freenove 4WD Smart Car Kit, if that adds anything.
Could I get any help?

Gadgetoid, 2023-03-06:
You're missing a comma right after `.hwver = 0xa02080`

Gadgetoid, 2023-03-06:
Note: this revision is missing from the project, so a PR including it would be welcome.

Jamie-Holland, 2023-03-06:
Ah right, thank you for pointing that out. I figured this revision wasn't supported so I ended up just upgrading to the RPi 4B anyways. It would be nice to see it included though, as it would make that Pi worth having, yet I understand its an uncommon revision so it really wouldn't be worth it. Thank you again though.

Gadgetoid, 2023-03-06:
I'm hoping we can make revision tracking a thing of the past, since there are just too many and any pretence that they're vetted or tested has been lost for years now. But yes, will flag this as a bug so it gets added.

clnorris, 2023-10-09:
@Gadgetoid Hi there, I am running into the similar issue getting a -3 error code on my board.  I purchased my RPi from a company selling a hat for FPP (Hansen Electronics).  At first it seemed strange that FPP would reject an RPi bought straight from the FPP supplier, but I guess there is a lot of variation among these RPi's.  Here is my cpuinfo, I am not sure of the status of this bug but would love to see this variant added.  Do I need to fork and PR, or is this enough description?  I need this rather soon, so may have to rebuild the fppd and rpi_ws281x while waiting on the official releases.

Hardware        : BCM2835
Revision        : a020d4
Serial          : 00000000cb616ed1
Model           : Raspberry Pi 3 Model B Plus Rev 1.4

xenoputtss, 2024-01-04:
> EDIT: Nevermind, I am seeing a020d4 in a later commit on master. I think I need to PR on FPP to use the more recent rpi_ws281x.
> 
> @Gadgetoid Hi there, I am running into the similar issue getting a -3 error code on my board. I purchased my RPi from a company selling a hat for FPP (Hansen Electronics). At first it seemed strange that FPP would reject an RPi bought straight from the FPP supplier, but I guess there is a lot of variation among these RPi's. Here is my cpuinfo, I am not sure of the status of this bug but would love to see this variant added. Do I need to fork and PR, or is this enough description? I need this rather soon, so may have to rebuild the fppd and rpi_ws281x while waiting on the official releases.
> 
> Hardware : BCM2835 Revision : a020d4 Serial : 00000000cb616ed1 Model : Raspberry Pi 3 Model B Plus Rev 1.4

I just discovered that this was my issue also.  I purchased a Raspberry Pi 3 B and I get the hardware not supported error.

my `cat /proc/cpuinfo` produces this

```
Hardware        : BCM2835
Revision        : a020d4
Serial          : 00000000b16000e0
Model           : Raspberry Pi 3 Model B Plus Rev 1.4
```

clnorris, 2024-01-04:
@xenoputtss If you are running FPP, the latest v7.4 is now submoduling a newer revision of rpi_ws281x that recognizes the newer RPi hardware.  I haven't tried it yet, but if you are running FPP it looks like an upgrade to v7.4 will provide the fix.

BTW, the source code of FPP including the rpi_ws281x is present in the FPP image, so it is also possible to patch the source in place on the target HW and and rebuild.  That's what I did in the fall to get by until my PR to upgrade rpi_ws281x in FPP was accepted.  Now that v7.4 is released, I'm going to try it again myself soon.

xenoputtss, 2024-01-04:
> @xenoputtss If you are running FPP, the latest v7.4 is now submoduling a newer revision of rpi_ws281x that recognizes the newer RPi hardware. I haven't tried it yet, but if you are running FPP it [...]

I don't know what FPP is.  I was attempting to user HyperBain (Hyperion on Pi).

Ill look into seeing how to rebuild this library with the changes needed to get this to work.  Maybe Ill learn what FPP is along that route.

clnorris, 2024-01-04:
@xenoputtss Falcon Player schedules and plays light show sequences, such as Christmas lights on a house.  I took a quick peek at Hyperion.  They have their own fork of the rpi_ws281x project, so you will actually want to be looking at https://github.com/hyperion-project/rpi_ws281x instead of this one.  It looks like the master branch of Hyperion has the RPi fix submoduled.  I am not familiar with Hyperion at all or how often they build, but if you could get an image cut from master then I think you would be good to go.  For Hyperion, this is the commit that has the fix.  https://github.com/hyperion-project/rpi_ws281x/commit/c09fecb5ead9a33696f78913627c5693ca6a5c5a