jgarff/rpi_ws281x #515: Compute Module 4 latest version not supported

gecastan, 2023-03-03:
Starting a basic test using rpi_ws281x lib raising  `ws2811_init failed with code -3 (Hardware revision is not supported)` error :

```bash
python built-in-rgb-led-simple.py
This examples needs to be run as root (sudo)
Traceback (most recent call last):
  File "/home/pi/piunora-blinka-examples/built-in-rgb-led-simple.py", line 22, in <module>
    strip.begin()
  File "/home/pi/.local/lib/python3.9/site-packages/rpi_ws281x/rpi_ws281x.py", line 131, in begin
    raise RuntimeError('ws2811_init failed with code {0} ({1})'.format(resp, str_resp))
RuntimeError: ws2811_init failed with code -3 (Hardware revision is not supported)
```

Here is my brand new compute module 4 received this week :

 ```bash
cat /proc/cpuinfo
Hardware        : BCM2711
Revision        : b03141
Serial          :  XXXX
Model           : Raspberry Pi Compute Module 4 Rev 1.1
 ```

Quickly browsed the repository and found out that the hardware check is done here : [rpi_ws281x](https://github.com/jgarff/rpi_ws281x/blob/master/rpihw.c) and couldn't find a reference to the `b03141` revision.

I m unfamiliar with this kind of stuff so I ll play around and test a bit to test it locally before submiting a MR.

creatron, 2023-03-06:
I had the same error. I use the 'C' version and just had to add this to "rpihw.c", and now everything seems to be working.
    //
    // Compute Module 4
    //
    {                     /// gerrie added this
        .hwver = 0xb03141,
        .type = RPI_HWVER_TYPE_PI4,
        .periph_base = PERIPH_BASE_RPI4,
        .videocore_base = VIDEOCORE_BASE_RPI2,
        .desc = "Compute Module 4 v1.? eMMC Gerrie"
    },

Gadgetoid, 2023-03-06:
Merged a bumper PR of compute modules that should sort these out.

closed 2023-03-06 by Gadgetoid